slviajero/tinybasic #12: Wrong tokenization or whitespace removal for if/then

jnorre, 2022-04-29:
Shortest example:

> 10 if 20>10 then goto 20
> list
10 IF 20>10THEN GOTO 20
> run
10: Error

Space after '20>10' has been removed and causes an error when the program is run. The same problem is also present when using variables (just used constants to keep it a one-liner).
I am using an Arduino UNO and has made the following two changes in hardware-arduino.h:
#undef USESPICOSERIAL 
#define ARDUINOEEPROM

slviajero, 2022-04-29:
Which code do you use? Is it the code in TinybasicArduino or the code in one of the other folders (the code that comes in 1 file)?

slviajero, 2022-04-29:
I checked the code and there is no tokenisation issue in any of the code parts. Whitespaces are removed in the older versions of the tokeniser but this doesn't cause a syntax error. If you compiled only the Palo Alto BASIC then the reason for the error is that THEN is not a keyword in Palo Alto. Try 
if 20>10 goto 10 
instead of 
if 20>10 then goto 10
To have THEN you need to compile with #define HASAPPLE1. For an Arduino UNO I recommend to set the options in the main code TinybasicArduino/TinybasicArduino.ino to compile #define BASICINTEGER. This is a integer BASIC with the full language set. It also has THEN and in the newest version also ELSE.

closed 2022-04-29 by slviajero

jnorre, 2022-04-30:
Thank for your feedback.
I tried it on another Arduino UNO on my own Ubuntu computer and had no problems - both with and without THEN. So it must be another installed library or board support library on my Windows work computer that messes up things.
Just a note: Default in TinybasicArduino is #define BASICINTEGER and hence #define HASAPPLE1, so THEN should be supported by a default build.

slviajero, 2022-04-30:
That is very odd! Can you tell me which IDE version? Need to check this. 
And yes, #define BASICINTEGER is with HASAPPLE1. I tried to stay as close to the original Palo Alto and did’t include THEN in the minimal language set. I loved to see how much one can do with so little.

jnorre, 2022-05-01:
It's the latest version 1.8.19 for Windows - I updated just before trying this out. On my Ubuntu, where it works, I am using v.1.8.15.

slviajero, 2022-05-01:
OK, great, thank you! I compile with 1.8.15 and test against the newest version 2.x and Arduino studio. I will check on Windows as soon as I have time. Busy preparing the 1.3 release of BASIC right now. The sensor library and display code needs cleaning up.

jnorre, 2022-05-02:
I checked again when I got back to my job today: The version of Tinybasic I downloaded Friday, has #define BASICMINIMAL as default, while the one I downloaded at home Saturday has #define BASICINTEGER defined as default. So it behaves like you have described, depending on which of these defines are active.

slviajero, 2022-05-07:
That was my mistake then. I accidentally have checked in with the wrong settings.