slviajero/tinybasic #13: Extending the language

Kongduino, 2022-05-10:
I got a bit lost between the tokens and the negative numeric values... Do you have a tutorial explaining how to extend the language?

Thanks

slviajero, 2022-05-22:
https://github.com/slviajero/tinybasic/wiki/How-to-compile-Stefan's-BASIC-(and-how-it-works)#how-to-add-your-own-commands

slviajero, 2022-05-22:
Tutorial added to documentation.

closed 2022-05-22 by slviajero

slviajero, 2022-10-11:
It is complicated. ;-) Arduino PROGMEM logic makes it a bit clumsy. I have written a few words at the end of this text right now: 
https://github.com/slviajero/tinybasic/wiki/How-to-compile-Stefan's-BASIC-(and-how-it-works) <https://github.com/slviajero/tinybasic/wiki/How-to-compile-Stefan's-BASIC-(and-how-it-works)>
Have a look and please tell me if this works for you. I can write more on specific topics if needed.