slviajero/tinybasic #17: ECMA55 test with numebrs beginning with a "." fail

EncomLab, 2022-06-18:
Any number starting with a decimal point causes a syntax error.   For example PRINT 2+.2 causes a syntax error.

EncomLab, 2022-06-18:
Further testing shows that this can be avoided by using a leading "0".  Probably not worth fixing but maybe worth a note in the documentation.

closed 2022-06-18 by EncomLab

slviajero, 2022-06-18:
Thank you for the comment. Yes, indeed something that should be changed. It really comes because I extended integer to float and simply did this by adding a . and then another number and then E and again another number to the grammar.

slviajero, 2022-06-18:
Corrected and pushed. The code can now parse .2 correctly in input and commands. 
Side effects not fully tested. Little oddity a=. assigns 0 to a now because . is seen as a number.

EncomLab, 2022-06-18:
I am working my way through all 208 ECMA55 test cases and will share the spreadsheet of my findings.  I am running the software on an ATMega 1284 SBC with an SD card, DS3231 rtc and 24C256 external eeprom.

slviajero, 2022-06-18:
Super cool, actually I wanted to build a 1284 board myself. 
As to the ECMA55 tests - I expect many to fail. Have done too many evil things.
Thanks again!