slviajero/tinybasic #21: GET broken (when using fabgl)

pgrobelniak, 2022-06-26:
Checking for break character seems to be messing with GET which always returns. Disabling it seem to help, but i can no longer stop my programs :)

https://github.com/slviajero/tinybasic/commit/de132c87c8f9e9c8f5518a44cd1f0ec39c7f34d7

slviajero, 2022-06-26:
> de132c8 <https://github.com/slviajero/tinybasic/commit/de132c87c8f9e9c8f5518a44cd1f0ec39c7f34d7>

Actually, on Arduino GET is meant to return. This is done by design. It is a non blocking stream input that will return 0 if no character is there.

The way to do blocking single character return from the serial stream is to write 

100 IF AVAIL(1) THEN GET A ELSE 100

i.e. you can check AVAIL on the stream if there is a character. AVAIL works for any input stream that supports it low level.

An elegant way to do it could be to use the special variables. So, equivalently you can write

100 IF @A THEN ***@***.*** ELSE 100

@A is AVAIL on the default stream and @C is GET on the default stream.

On Posix GET will block because there it is build on getchar which blocks. There is no simple portable non blocking IO on POSIX.

pgrobelniak, 2022-06-26:
Wow, thanks for such quick response :) I'm trying to get a character from ps2 keyboard using fabgl lib on esp32. Using this

10 if avail(2) then get a else 10

results in "10: Syntax error" for some reason. 

Also i have to press a key few times to make it work for some reason.

Great project btw :)

slviajero, 2022-06-26:
Too bad, you found an else bug. I added this only recently and it still has some flaws

Use this instead

10 if avail(2) then get a: goto 30
20 goto 10
30 print a

Tried this on an ESP32 right now (I actually work on MQTT at the moment). I will check to ELSE thing later.

slviajero, 2022-06-26:
Bug is fixed now. Please download the newest code basic.c or TinybasicArduino.ino  file.
This was a bug in the token stream handling of GET which only matters if it comes with ELSE.
I switched to strict syntax checking a week ago which uncovers a few hidden bugs.

pgrobelniak, 2022-06-26:
The code works now but i always get 0. My theory is that kbdcheckch with PS2FABLIB "consumes" the character before i can get it

slviajero, 2022-06-26:
Ok, I check this later in the evening. Got a fab32 in my office.

slviajero, 2022-06-26:
I checked it again and you were of course right. The way kbdcheckch() is implemented it swallows characters every time a BREAKCHAR is checked. Unfortunately Terminal.peek() doesn’t seem to work. I tried it right now and there seems to be an issue. It always returns 255 and cannot be used to check the stream. This may be connected to the VT52 terminal used in BASIC.

I implemented a workaround in hardware-arduino.h that can do BREAKCHAR and GET.  Characters checked are put into a one character buffer and are then reinserted into the stream.

It has the same weakness as the serial code. If you send some characters and then BREAKCHAR the program does not stop.

pgrobelniak, 2022-06-26:
Fantastic, it works now. Thanks!

closed 2022-06-26 by pgrobelniak

slviajero, 2022-10-11:
PS: there are a few examples here: https://github.com/slviajero/tinybasic/tree/main/examples/04communication <https://github.com/slviajero/tinybasic/tree/main/examples/04communication>
on AVAIL and GET