slviajero/tinybasic #24: Soundgenerator in TTGOVGA 1.4

testerrossa, 2022-07-08:
Hi, i did not want to fork the whole thing, but if you want you can add sound in the TTGOVGA 1.4 very easily with the FabGL-functions like this, this is a modified version of btone in hardware-arduino.h. With this mod you can play music:

play (waveform, frequency, duration)
waveform beeing: 
1 sine
2 square
3 sawtooth
4-8 triangle
9-125 squarewave with different dutycycles
126 and 127 two different noisegenerators

void btone(short a) {
  number_t d = 0;
  if (a == 3) d=pop();
	x=pop();
	y=pop();
#if defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_ESP32)
#if defined(ARDUINO_TTGO_T7_V14_Mini32)
   if (x == 0) {
    //noTone(y);
    soundGenerator.play(false);
  } else if (a == 2) {
    // tone(y, x); pin,frequency 
    // soundGenerator.playSound (waveform, frequency, durationMS, volume) ,  Minimum for vol is 0, maximum is 127.
    if (y==1) soundGenerator.playSound(SineWaveformGenerator(), x, 60000);    //sine
    if (y==2) soundGenerator.playSound(SquareWaveformGenerator(), x, 60000);  //square
    if (y==3) soundGenerator.playSound(SawtoothWaveformGenerator(), x, 60000);  //sawtooth
    if (y>=4) {
      if (y<=8) soundGenerator.playSound(TriangleWaveformGenerator(), x, 60000); //triangle ...
    } 
    if (y>=9) { 
      SquareWaveformGenerator sqw;
      sqw.setDutyCycle(y*2);
      if (y<=125) soundGenerator.playSound(sqw, x, 60000); //square ...
    }
    if (y==126) soundGenerator.playSound(VICNoiseGenerator(), x, 60000);    //VIC noise
    if (y==127) soundGenerator.playSound(NoiseWaveformGenerator(), x, 60000); //noise
    
  } else {
    if (y==1) soundGenerator.playSound(SineWaveformGenerator(), x, d);    //sine
    if (y==2) soundGenerator.playSound(SquareWaveformGenerator(), x, d);  //square
    if (y==3) soundGenerator.playSound(SawtoothWaveformGenerator(), x, d);  //sawtooth
    if (y>=4) {
      if (y<=8) soundGenerator.playSound(TriangleWaveformGenerator(), x, d); //triangle ...
    } 
    if (y>=9) { 
      SquareWaveformGenerator sqw;
      sqw.setDutyCycle(y*2);
      if (y<=125) soundGenerator.playSound(sqw, x, d); //square ...
    }
    if (y==126) soundGenerator.playSound(VICNoiseGenerator(), x, d);    //VIC noise
    if (y==127) soundGenerator.playSound(NoiseWaveformGenerator(), x, d); //noise
    //tone(y, x, d);   y = pin, x =freq , d = duraton
  }
  #endif
	return;
#else 
  if (x == 0) {
    noTone(y);
  } else if (a == 2) {
	tone(y, x);
	} else {
	tone(y, x, d);
	}
#endif	
}

For me I even extend the play command with a 4th parameter for volume.

cheers

slviajero, 2022-07-08:
Cool! I will integrate this into the main code. If you give me the 4 parameter version as well, I will also integrate this. 
I had plans to write sound code but it was not on my top priority list.

Currently  have a few old 8bit sound chips (AY-3-8910) that I will write a driver for.

Best, Stefan

testerrossa, 2022-07-08:
here is the Version with volume control

in TinybasicArduino.ino change to allow for a 4th parameter

/* tone if the platform has it -> BASIC command PLAY */
#ifdef HASTONE
void xtone(){
	nexttoken();
	parsearguments();
	if (er != 0) return;
	if (args>4 || args<2) {
		error(EARGS);
		return;
	}
	btone(args);	
}
#endif

in hardware-arduino.h:

void btone(short a) {
  number_t d = 0;
  number_t v = 100;
  if (a == 4) v=pop();
  if (a >= 3) d=pop();
	x=pop();
	y=pop();
#if defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_ESP32)
#if defined(ARDUINO_TTGO_T7_V14_Mini32)
   if (x == 0) {
    //noTone(y);
    soundGenerator.play(false);
  } else if (a == 2) {
    // tone(y, x); pin,frequency 
    // soundGenerator.playSound (waveform, frequency, durationMS, volume) ,  Minimum for vol is 0, maximum is 127.
    if (y==1) soundGenerator.playSound(SineWaveformGenerator(), x, 60000,v);    //sine
    if (y==2) soundGenerator.playSound(SquareWaveformGenerator(), x, 60000,v);  //square
    if (y==3) soundGenerator.playSound(SawtoothWaveformGenerator(), x, 60000,v);  //sawtooth
    if (y>=4) {
      if (y<=8) soundGenerator.playSound(TriangleWaveformGenerator(), x, 60000,v); //triangle ...
    } 
    if (y>=9) { 
      SquareWaveformGenerator sqw;
      sqw.setDutyCycle(y*2);
      if (y<=125) soundGenerator.playSound(sqw, x, 60000,v); //square ...
    }
    if (y==126) soundGenerator.playSound(VICNoiseGenerator(), x, 60000,v);    //VIC noise
    if (y==127) soundGenerator.playSound(NoiseWaveformGenerator(), x, 60000,v); //noise
    
  } else {
    if (y==1) soundGenerator.playSound(SineWaveformGenerator(), x, d,v);    //sine
    if (y==2) soundGenerator.playSound(SquareWaveformGenerator(), x, d,v);  //square
    if (y==3) soundGenerator.playSound(SawtoothWaveformGenerator(), x, d,v);  //sawtooth
    if (y>=4) {
      if (y<=8) soundGenerator.playSound(TriangleWaveformGenerator(), x, d, v); //triangle ...
    } 
    if (y>=9) { 
      SquareWaveformGenerator sqw;
      sqw.setDutyCycle(y*2);
      if (y<=125) soundGenerator.playSound(sqw, x, d,v); //square ...
    }
    if (y==126) soundGenerator.playSound(VICNoiseGenerator(), x, d, v);    //VIC noise
    if (y==127) soundGenerator.playSound(NoiseWaveformGenerator(), x, d, v); //noise
    //tone(y, x, d);   y = pin, x =freq , d = duraton
  }
  #endif
	return;
#else 
  if (x == 0) {
    noTone(y);
  } else if (a == 2) {
		tone(y, x);
	} else {
		tone(y, x, d);
	}
#endif	
}

that's it 

cheers from Grasberg bei Bremen

slviajero, 2022-07-08:
Thank you! You probably needs something like 

#include <soundgen.h> 

and a

SoundGenerator soundGenerator;

somewhere in the code to create the necessary object.

testerrossa, 2022-07-08:
ups,
totally forgot about that.

it is in hardware-arduino.h after the include for fabgl

/*
 * VGA is only implemented on one platform - TTGO VGA 1.4
 * This does currently not compile with the newest ESP32 
 * Tested with 
 * https://github.com/slviajero/FabGL
 * Newer versions not yet tested
 */
#if defined(ARDUINOVGA) && defined(ARDUINO_TTGO_T7_V14_Mini32)
#include <WiFi.h> 
#include <fabgl.h> 
SoundGenerator       soundGenerator;
#endif

slviajero, 2022-07-09:
Integrated the code right now with a tiny change, I use pin numbers 128 upward to address the sound generator capabilities. Square wave duty cycle is addressed from 256 upward. This way the BASIC programs using PIN based tone and the sound generator remain compatible. 

Problem is that I could not test it properly as the sound output of my board is broken. Code seems to run. If you find time you might test it and give me feedback.

testerrossa, 2022-07-09:
Hi,
just compiled it and it works perfectly.

thanks

slviajero, 2022-07-09:
Good! Thank you for the contribution!

slviajero, 2022-07-12:
Code integrated to main code.

closed 2022-07-12 by slviajero