slviajero/tinybasic #26: How to return a factor to a variable

EncomLab, 2022-07-18:
My goal is to assign a variable to a value returned from a sensor - for example:
```
 10 LET U=PING 1,2
 20 PRINT U
```
U = cm where PING is:
```
void xping() {
  nexttoken();
  parsenarguments(2);

  //if (er != 0) return;
  x = pop();
  y = pop();

  short int trigPin; //required
  short int echoPin;
  long duration, cm; //inches;

  trigPin = pinNum(y);
  echoPin = pinNum(x);

  //initialize ping
  pinMode(trigPin, OUTPUT);//trigger
  pinMode(echoPin, INPUT);//echo

  // The sensor is triggered by a HIGH pulse of 10 or more microseconds.
  // Give a short LOW pulse beforehand to ensure a clean HIGH pulse:
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);

  // Read the signal from the sensor
  pinMode(echoPin, INPUT);
  duration = pulseIn(echoPin, HIGH);

  // Convert the time into a distance
  cm = (duration / 2) / 29.1;   // Divide by 29.1 or multiply by 0.0343

  Serial.print(cm);
  push(cm);
}
```

This prints the correct distance but I do not understand the steps required to have the result assignable to a variable.

slviajero, 2022-07-18:
xping has to be a function. This means it will be something like 

10 LEN U=PING(1,2)

so it is subscripts and not arguments you want to parse. There is a method already implemented for 
it. Most of your function can stay but it has to be hooked into the factor function. Let’s look at an example, 
the USR function. 

Look for the code 

		case TUSR:
			parsefunction(xusr, 2);
			break;

This is how a function is hooked into the factor routine. It handles all the integration into the arithmetic.
You need to add your token and function here and not in statement().

Then look at the function xusr itself. It simply pops two arguments and pushes one. There is no nexttoken or 
anything else required. parsefunction() does that for you. 

The correct code would be 

In factor 

	case TPING:
		parsefunction(xping, 2); // parse a function with two arguments
		break;

and then somewhere else

> void xping() {
> 
>   x = pop();
>   y = pop();
> 
>   short int trigPin; //required
>   short int echoPin;
>   long duration, cm; //inches;
> 
>   trigPin = pinNum(y);
>   echoPin = pinNum(x);
> 
>   //initialize ping
>   pinMode(trigPin, OUTPUT);//trigger
>   pinMode(echoPin, INPUT);//echo
> 
>   // The sensor is triggered by a HIGH pulse of 10 or more microseconds.
>   // Give a short LOW pulse beforehand to ensure a clean HIGH pulse:
>   digitalWrite(trigPin, LOW);
>   delayMicroseconds(5);
>   digitalWrite(trigPin, HIGH);
>   delayMicroseconds(10);
>   digitalWrite(trigPin, LOW);
> 
>   // Read the signal from the sensor
>   pinMode(echoPin, INPUT);
>   duration = pulseIn(echoPin, HIGH);
> 
>   // Convert the time into a distance
>   cm = (duration / 2) / 29.1;   // Divide by 29.1 or multiply by 0.034
>   push(cm);
> }

Best Regards,
Stefan

slviajero, 2022-07-18:
> Works perfectly - thanks again.

Enjoy! Pleasure to help.