slviajero/tinybasic #28: Missing (important) detail in "How to add your own commands"

Kongduino, 2022-08-24:
In [How to add your own commands](https://github.com/slviajero/tinybasic/wiki/How-to-compile-Stefan's-BASIC-(and-how-it-works)#how-to-add-your-own-commands) I think you skipped an important step, which is what tripped me while trying to add a command.

In basic.h I see these defines:

```c
/* the number of keywords, and the base index of the keywords */
#define NKEYWORDS	3+19+13+12+11+5+2+7+7+7+12
#define BASEKEYWORD -121
```

Shouldn't `NKEYWORDS` be incremented to match the number of total keywords with the one(s) we add? I don't see any mention of this anywhere.

Thanks.

slviajero, 2022-08-24:
Until version 1.2. this was needed. Now it is not used any more. I keep it updated more out of tradition to know how many keywords I have added to each language set.

Kongduino, 2022-08-25:
Ah thanks, I was still on v1.2!

closed 2022-08-25 by Kongduino