slviajero/tinybasic #30: Problems compiling for Arduino Uno

blogmywiki, 2022-09-08:
I'm having problems compiling for the Arduino Uno as  the target board. I keep getting the following error, regardless of what options I have defined or undefined. I'd be very grateful if anyone can shed any light on what's going wrong.

```
In file included from /Users/xxx/Documents/Arduino/TinyBasic/TinybasicArduino/TinybasicArduino.ino:212:0:
/Users/xxx/Documents/Arduino/TinyBasic/TinybasicArduino/hardware-arduino.h: In function 'int freeRam()':
hardware-arduino.h:653:5: error: ambiguating new declaration of 'int freeRam()'
 int freeRam() {
     ^~~~~~~
In file included from /Users/xxx/Documents/Arduino/TinyBasic/TinybasicArduino/TinybasicArduino.ino:200:0:
/Users/xxx/Documents/Arduino/TinyBasic/TinybasicArduino/basic.h:843:6: note: old declaration 'long int freeRam()'
 long freeRam();
      ^~~~~~~
exit status 1
ambiguating new declaration of 'int freeRam()'
```

pgrobelniak, 2022-09-08:
Just change int to long in hardware -arduino.h

blogmywiki, 2022-09-08:
Thank you - that worked!

closed 2022-09-08 by blogmywiki

slviajero, 2022-09-09:
This is a bug in v13. I fixed it in the meantime but have not yet commited the change. Will do this today. Pavels comment is correct, it should be long everywhere in the code.

slviajero, 2022-09-09:
Fixed in the repo, freeRam is now long everywhere.

pgrobelniak, 2022-10-11:
Just change the int to long in hardware-arduono.h

slviajero, 2022-10-11:
Also changed in my repo a week ago …
Was a glitch.