slviajero/tinybasic #31: While you are working on ISR

EncomLab, 2022-10-18:
Morning Stefan! The other way in which I am currently using interrupts for the robot platform is to handle the output from optical rotary encoders.  I currently just use this to get a step count by assigning a ENCODERA or ENCODERB command to a variable and assigning the encoder a pin number.  Then I just use the variable in an IF statement to issue a STOP command to the motors.  

`//MOTOR AND ENCODER FUNCTIONS
//soft reset
void interruptFunctionSR() {
  xbreak();
}

//encoder a
volatile uint16_t interruptCountA = 0;
void interruptFunctionA() {
  interruptCountA++;
}

void xencodera() {
  x = pop();
  short int encoderaPin;
  encoderaPin = pinNum(x);

  pinMode(encoderaPin, INPUT_PULLUP);
  enableInterrupt(encoderaPin, interruptFunctionA, CHANGE);

  push(interruptCountA);

}

//encoder b
volatile uint16_t interruptCountB = 0;
void interruptFunctionB() {
  interruptCountB++;
}

void xencoderb() {
  x = pop();
  short int encoderbPin;
  encoderbPin = pinNum(x);

  pinMode(encoderbPin, INPUT_PULLUP);
  enableInterrupt(encoderbPin, interruptFunctionB, CHANGE);

  push(interruptCountB);

}`

slviajero, 2022-10-19:
Looks very good. That is exactly following the logic of the interpreter. I need to think about how to generalise this. BASIC is not the ideal language to handle interrupts because there is no way to define procedures / call backs.

slviajero, 2023-02-05:
Just for info, I now have released the first code with built-in interrupt code. It is documented in the MANUAL, the command EVENT is doing the interrupt stuff. Still working to improve it.

closed 2023-02-05 by slviajero