slviajero/tinybasic #32: Can not compile for TTGO VGA 1.4

Technikfreak2002, 2022-10-22:
Hello! I found your amazing project and give it a try. I want to compile for TTGO VGA 1.4. I found all lib's and tested with a FabGl example. all fine.
I followed your instructions but i get some erros. if i use your predefined TTGVGA board switch i get the same erros.
i used a fresh arduino 1.8.19 installation as well as the new 2.0 IDE with the same results.
"ESP32 Dev Module" with PSRAM disabled and default partion scheme. 
Gives me the following errors:
`/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z5xplotv+0x0): undefined reference to `plot(int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z5xlinev+0x0): undefined reference to `line(int, int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z5xrectv+0x0): undefined reference to `rect(int, int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z7xcirclev+0x0): undefined reference to `circle(int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z6xfrectv+0x0): undefined reference to `frect(int, int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z8xfcirclev+0x0): undefined reference to `fcircle(int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z6xcolorv+0x0): undefined reference to `vgacolor(short)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o:(.literal._Z6xcolorv+0x4): undefined reference to `rgbcolor(int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xplot()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5574: undefined reference to `plot(int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xline()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5590: undefined reference to `line(int, int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xrect()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5602: undefined reference to `rect(int, int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xcircle()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5613: undefined reference to `circle(int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xfrect()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5625: undefined reference to `frect(int, int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xfcircle()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5636: undefined reference to `fcircle(int, int, int)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino-sketch-E5F4A97600AA32C4BFD77415031CBB34/sketch/TinybasicArduino.ino.cpp.o: in function `xcolor()':
/home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5551: undefined reference to `vgacolor(short)'
/home/markus/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/markus/Arduino/tinybasic/TinybasicArduino/TinybasicArduino.ino:5556: undefined reference to `rgbcolor(int, int, int)'
collect2: error: ld returned 1 exit status
Mehrere Bibliotheken wurden für "SD.h" gefunden
  Benutzt: /home/markus/.arduino15/packages/esp32/hardware/esp32/2.0.5/libraries/SD
  Nicht benutzt: /home/markus/Arduino/libraries/SD
exit status 1

Compilation error: exit status 1`

All lib's and boards are up to date. 
Any ideas?
Thank you very much!

slviajero, 2022-10-22:
Esp 32 Dev module is the wrong board.
Try the TTGO VGA 1.4 Board further down in the boarss menu.

Technikfreak2002, 2022-10-22:
Yes i searched for a VGA board. but there isn't such a board. So my choice was the ESP32 dev board. my i have to install a different package for that?

slviajero, 2022-10-22:
Sorry, my mistake, the board is of course the TTGO T7 V1.4 mini32
No mention of VGA. 
This is the base board they used to make the VGA kit wit.

I just checked and compiled.

Technikfreak2002, 2022-10-22:
Yes confirmed it is working! Thank you very much!

closed 2022-10-22 by Technikfreak2002

slviajero, 2022-10-22:
Enjoy!