slviajero/tinybasic #33: External EEPROM as program storage

EncomLab, 2022-11-07:
Stefan can you speak further to the use of an external EEPROM - specifically an EEPROM module utilizing the 24C256 - to expand the primary program space?

slviajero, 2022-11-07:
WIll do! Give me time till weekend, currently I am very busy with my day job. I have a subsystem for serial RAM chips and options for EEPROM use ready.

slviajero, 2022-11-10:
I wrote a few words now by updating the wiki article 

https://github.com/slviajero/tinybasic/wiki/SD-Cards,-SPIFFS,-File-commands,-EEPROM,-and-I-O-channels.

In brief words: External EEPROM can be connected with the ARDUINOI2CEEPROM option in the code now.  I added this only recently. You can save a program to the external EEPROM and then run it from there. Using SET 1,1 you can set it to autorun. Autorun modules start on an Arduino without loading the program to RAM. It runs purely from EEPROM and the entire memory is free for variables. 

The fun thing about it is that you can create the module on an Arduino with a lot of memory, like a MEGA256 and then use it on an UNO as the modules and the code is interchangeable. This way you can run 8kB programs on an 1 kB UNO. 

I so far have not added the possibility to edit programs stored in EEPROM. You need to create and edit the program in RAM, then save it to EEPROM and then run it from there with the autorun feature. 

Adding editing of the EEPROM code is not super difficult but will be very slow. 

Let me know you thoughts on that! What would be the use case you have in mind?

By the way: there is some info on connecting EEPROMs in general here: https://create.arduino.cc/projecthub/sl001/use-i2c-eeproms-as-a-file-system-on-an-arduino-c8f2a2 <https://create.arduino.cc/projecthub/sl001/use-i2c-eeproms-as-a-file-system-on-an-arduino-c8f2a2>

A further, yet undocumented feature is that you can use 23C512 modules as RAM extension. This is serial SPI SRAM. 32KB or 64 kB RAM is possible with these modules. This is the ARDUINOSPIRAM option in the code.

slviajero, 2022-11-19:
I have now added true EEPROM program support as an experimental code. 

Currently it is only checked in here: https://github.com/slviajero/tinybasic/tree/main/TinybasicArduino <https://github.com/slviajero/tinybasic/tree/main/TinybasicArduino>

With the setting 

#define ARDUINOPGMEEPROM

all the EEPROM memory is used for program storage and can be changed interactively. You have practically EEPROM size + Memory size as BASIC memory. I added limits internally so that program can only be in EEPROM and variables only in RAM to avoid noodling the EEPROM to death in a loop. 

The code has some limits and is very experimental. I only tested it with the internal EEPROM of an UNO. It should work with an external EEPROM but I didn’t test it so far. 

The main limit is that the string length can only be max 128 bytes. So, no more arbitrary length strings. This is because I need two buffers in true RAM to copy strings around.

EncomLab, 2022-11-19:
This is amazing - the ability to use a cheap Uno as the interface to a larger memory capacity system is a game changer.

slviajero, 2022-11-20:
Thank you! Please be careful with the code. It is test code, there will be many bugs and unwanted side effects. Will take me a week to find them and fix them.

slviajero, 2022-11-20:
Code is clean now in beta state in the folder https://github.com/slviajero/tinybasic/tree/main/TinybasicArduino

Please download the newest version of EepromFS from 
https://github.com/slviajero/EepromFS
if you want to use it with I2C EEPROMS

Limitations: 
- RAM and EEPROM together cannot be more than 32kB on an integer BASIC system. Limit the EEPROM size parameters accordingly, on a float system there are no limitations
- String lengths is limited to 128 bytes as the buffer code from SPIRAM is used
- EEPROM are fully compatible with EEPROMs generated by SAVE and LOAD to EEPROM targets as long as the number_t and address_t are the same i.e. you can exchange EEPROMs between float systems and between integer systems but an integer EEPROM will not work on a float system

closed 2022-11-20 by slviajero