slviajero/tinybasic #34: Various issues encountered

brentdcarlson, 2022-11-09:
Several issues I have encountered: 

run num doesn't work

@S doesn't report end of file for file read

open "filename",2 - doesn't append to an existing file

delete "filename" doesn't work

display scrolling doesn't update colors properly when text is not updated

slviajero, 2022-11-10:
Thank you for the report. 

As to RUN num: this should work but doesn’t any more. Probably a bug coming from the command line code. You can use GOTO num from the command line as a workaround. GOTO num doesn’t CLR but starts the program. I will look into this.

The filesystem related things are probably coming from the filesystem implementation of the platform. The SD libraries of different cores are often very different. If I have the e xact configuration I can rebuild it. Can you give me more details, please? Which platform and which hardware-settings? I would need the hardware-arduino.h hardware settings file. 

Display: The reason for this is that the display driver is not really color aware. It only has a text buffer. This is because the code does not hardware scroll but redraw the text instead. It is a works as designed type of thing. Never thought of it. Will look into it now as I am about to rewrite the display driver.

slviajero, 2022-11-10:
Run NUM is fixed now for all platforms

brentdcarlson, 2022-11-10:
Thanks for the quick response.  I'm enjoying working with your platform.

Using Arduino IDE 2.0.1, "DOIT ESP32 DEVKIT V1", 3.5" display, ILI9488 controller with SD card for file storage, BMP280, DHT22, DS3231 RTC

#undef USESPICOSERIAL 
#define ARDUINOPS2
#undef ARDUINOUSBKBD
#undef ARDUINOZX81KBD
#undef ARDUINOPRT
#define DISPLAYCANSCROLL
#undef ARDUINOLCDI2C
#undef ARDUINONOKIA51
#define ARDUINOILI9488
#undef ARDUINOSSD1306
#undef ARDUINOMCUFRIEND
#undef ARDUINOGRAPHDUMMY
#undef LCDSHIELD
#undef ARDUINOTFT
#undef ARDUINOVGA
#undef ARDUINOEEPROM
#undef ARDUINOI2CEEPROM
#undef ARDUINOEFS
#define ARDUINOSD
#undef ESPSPIFFS
#undef RP2040LITTLEFS
#define ARDUINORTC
#define ARDUINOWIRE
#undef ARDUINOWIRESLAVE
#undef ARDUINORF24
#undef ARDUINOETH
#define ARDUINOMQTT
#define ARDUINOSENSORS
#undef ARDUINOSPIRAM 
#undef STANDALONE

slviajero, 2022-11-10:
Super, thank you! ESP32. I thought so. One of the platforms that is a little different. I could reproduce the EOF bug. Looks like ESP32 does not return an EOF character at end of file. Will try to fix this now.

brentdcarlson, 2022-11-10:
I used:

230 IF(A$(1,1))=255 THEN PRINT "End of file reached"

slviajero, 2022-11-10:
Just working on it. Looks like the ESP32 FS.cpp which is the base class of all the file systems on ESP somehow handles available() incorrectly but I could be wrong.

slviajero, 2022-11-10:
Found it! 

char in ESP32 is unsigned char hence 255 is EOF while in other platforms it is signed char hence -1 is EOF. 

In my BASIC char is always signed char, hence 255 = -1. This goes deep in the code. Need to take a careful look.

slviajero, 2022-11-10:
Bug fix is checked in @s and EOF is now handled correctly. I still left chars as unsigned chars on ESP.

slviajero, 2022-11-12:
ESP32 FS bugs are fixed now, delete works and also file append.

slviajero, 2022-11-12:
Issue closed, all bugs fixed except the scrolling part. This requires a redesign of the display driver code. It is to some degree "works as designed" as the display driver was meant to be monochrome from the beginning. Will add this to the documentation and then work on it. Will be a backlog item in the project.

closed 2022-11-12 by slviajero

slviajero, 2022-12-19:
Just a final remark on the issues: color support on scroll is now added. Limited to VGA colors to save buffer space.