slviajero/tinybasic #35: [request] use internal eeprom LittleFS filesystem for esp8266

atesin, 2022-11-16:
i noticed currently littlefs is only supported for raspberry pi pico :(
esp8266 has 2-4mb of flash that can be used for internal storage with spiffs or littlefs

https://arduino-esp8266.readthedocs.io/en/3.0.2/filesystem.html#spiffs-and-littlefs

slviajero, 2022-11-17:
This is true. Reason for it is that BASIC doesn’t need the features of littleFS. As there are only short and simple filenames and no folders SPIFFS was the most simple and widespread option. On RP2040 littleFS is the only option. I used the C style wrapper as the whole implementation is not really good on RP2040. 

Is there a particular reason, why you can’t use SPIFFS on ESP?

atesin, 2022-11-18:
> Is there a particular reason, why you can’t use SPIFFS on ESP?

because littlefs support true subdirectories and spiffs don't (they are emulated including slashes in root file names), littlefs is fault tolerant and spiffs is not, because spiffs is old and littlefs is new, and because spiffs development is inactive and is gonna be DEPRECATED... both are open source, i won't surprise if manufacturers start to drop spiffs to switch to littlefs

https://arduino-esp8266.readthedocs.io/en/3.0.2/filesystem.html#spiffs-deprecation-warning

slviajero, 2022-11-19:
Very true, but currently not a priority as BASIC really doesn’t need more than SPIFFS can offer.
The deprecation notice is there for years now but they still maintain it in the ESP32 core, even in the newest ones.
Will put that into the backlog for further development.

slviajero, 2024-01-26:
Taken as item into the project backlog. On hold until a contributor volunteers to implement it.

closed 2024-01-26 by slviajero