slviajero/tinybasic #36: Some examples not working, wrong usr() return value?#

stopnoanime, 2022-12-01:
Hi. First, thanks for all the work you put into this BASIC, it works well on a PI Pico.

But i have issues with launching some included examples, like lunar.bas. They check for the presence of floating point numbers by running usr(0,3).

This itself works fine, and the function returns 0 if there is no float, and 1 if there is float available.

The problem is that the examples NOT this value, changing 0 to -1, and 1 to -2. This causes incorrect logic behaviour and shows the message about there not being floating point support, on systems that have floating point support.

To clarify, this is what is happening:

| Before NOT | After NOT |
| ------------- | ------------- |
| 0 (false)  | -1 (true)  |
| 1 (true) | -2 (true) |

I don't know if I am missing something, or there is something wrong with the examples or the usr() return value.

Thanks

slviajero, 2022-12-02:
oh my! i changed the logic recently but not the examples ...

slviajero, 2022-12-02:
Hello Michal,

this bug comes from a late minute change I implemented changing from C style boolean to BASIC style boolean.

It was before 

NOT 0 = 1 
NOT 1 = 0

now it is 16 bit bitwise arithmetic with

NOT -1 = 0 
NOT 0 = -1

This made it possible to double use NOT, AND, OR as boolean and as bitwise operators. (Something Bill Gates already
did in the 70s). Unfortunately I forgot to change usr(0,3) respectively. I fixed this now and checked the change in. Should work now. 

If you want to fix it in your code without downloading the new version, simply change around line 6140 in the interpreter the 
section

#ifdef HASFLOAT
				case 3:	push(1); break;
#else 
				case 3: push(0); break;
#endif

to 

#ifdef HASFLOAT
				case 3:	push(-1); break;
#else 
				case 3: push(0); break;
#endif

Best,
Stefan

stopnoanime, 2022-12-02:
Thank you very much for the explanation and fixing the issue so quickly.

closed 2022-12-02 by stopnoanime