slviajero/tinybasic #40: Floating point reverting to integers?

RonLinu, 2022-12-11:
Hi Stefan,

In your last updated version of TinyBasic (when compiled as BASCIFULL or BASICTINYWITHFLOAT), 
the execution seems to revert to integers in some situations. In the previous version of TinyBasic, it was ok.

The simple Fibonaci program below can show the problem. When the loop reaches C=23, it begins to print
negative numbers instead of floating point format. When the loop reaches 36, it goes back to floating point
and stay that way.

20 A=0
30 B=1
60 FOR C=1 TO 40
70 F=A+B
90 A=B
100 B=F
105 PRINT C, F
110 NEXT C

Output results:
----------------------------
....                     ; from C=1 to 22, it's ok
22 28657
23 -19168        ; shouldn't print a negative number here
24 9489
25 -9679          ; again for the next six values...
26 -190
27 -9869         
28 -10059
29 -19928
30 -29987
31 15621
32 -14366
33 1255
34 -13111
35 -11856
36 2.41578E7             ; but it is ok from here...
37 3.90882E7
38 6.3246E7
39 1.02334E8
40 1.6558E8

slviajero, 2022-12-11:
Thank you! I messed this up during code cleanup. Fixed in the repo now. 
You can simply fix it by setting

typedef long wnumber_t;

in basic.h where the HASFLOAT thing happens. Line 638. 

> Am 11.12.2022 um 15:25 schrieb RonLinu ***@***.***>:
> 
> 20 A=0
> 30 B=1
> 60 FOR C=1 TO 40
> 70 F=A+B
> 90 A=B
> 100 B=F
> 105 PRINT C, F
> 110 NEXT C

RonLinu, 2022-12-11:
Thumb up for the super-quick response! :)

As you are updating the BASIC.H file, may I suggest changing the version number from 1.4a to something like 1.4b, just to identify the new version.

Cheers!

slviajero, 2022-12-13:
fixed the type in wnumber_t

closed 2022-12-13 by slviajero