slviajero/tinybasic #44: Allow the user to assign an autorun pin

twoporylyj, 2022-12-28:
Greetings. For now, we have the following in the manual:

"SET 1,1 activates the autorun mode of the EEPROM. SET 1,0 resets the autorun mode. SET 1,255 marks the EEPROM as not to contain a program. SET 1,1 should only be used if a program was stored with SAVE "!" to the EEPROM. There is no safety net here"

This problem can be solved by allowing the user to assign an autorun pin, so that program from EEPROM starts automatically only if that pin is grounded at startup or reset, otherwise the user gets a Basic interpreter prompt in the terminal and can edit and run the program manually. Thanks.

slviajero, 2022-12-28:
Thank you! Actually, this mechanism already exists in a slightly different form.  
In hardware-arduino.h there is the macro BREAKPIN. If you set that to any pin, 
then setting the pin to LOW will interrupt the program. The pin is set to 
INPUT_PULLUP at startup, no external additional circuitry is needed. I forgot 
to put that into the manual. Will do this asap.

twoporylyj, 2022-12-28:
Thanks, will try this.

slviajero, 2022-12-28:
Great, by the way, you can always interrupt a program by sending the BREAKCHAR. This is # by default. Can be changed to any character.

twoporylyj, 2022-12-28:
Cool!
10 print &2,"abcd"
20 goto 10
run
It began printing "abcd" on HD44780 in endless loop. Then i pressed # without even pressing enter, and program stopped. Thanks.

slviajero, 2022-12-28:
That’s how it is supposed to work. One remark: the break character has to be the first character send. If you type anything else and then # it won’t work. This is why I added the BREAKPIN. It works always.

twoporylyj, 2022-12-28:
But is case of error i can push restart button and send # again?

slviajero, 2022-12-28:
Yes! It is the statement loop that looks into the input stream. It checks the character in the buffer but does not read it. That’s why it has to be the first character. After reset, the buffer is empty. You  can go again.

closed 2022-12-29 by slviajero