slviajero/tinybasic #45: What is ZX81 keyboard?

twoporylyj, 2022-12-28:
Please tell me what is ZX81 keyboard in context of this interpreter? Is it a keyboard matrix wired like a real ZX81 keyboard? What are default pins for its connection? Can they be changed? Thanks.

slviajero, 2022-12-28:
https://create.arduino.cc/projecthub/sl001/read-a-zx81-keyboard-with-arduinos-and-build-things-with-it-0189bd <https://create.arduino.cc/projecthub/sl001/read-a-zx81-keyboard-with-arduinos-and-build-things-with-it-0189bd>

https://create.arduino.cc/projecthub/sl001/build-a-64kb-home-computer-based-on-an-arduino-mega-256-4aa798 <https://create.arduino.cc/projecthub/sl001/build-a-64kb-home-computer-based-on-an-arduino-mega-256-4aa798>

A real ZX81 Keyboard ...

twoporylyj, 2022-12-28:
Thanks. Are pins reassignable for nano, for example?

slviajero, 2022-12-28:
Yes. It is the section 

/* 
 *  Pin settings for the ZX81 Keyboard
 *  first the 8 rows, then the 5 columns or the keyboard
 * 
 * MEGAs have many pins and default is to use the odd pins on the side
 * UNOs, NANOs, and others use the lower pins by default avoiding the 
 * pin 13 which is LED and doesn't work with standard schematics 
 */
#ifdef ARDUINOZX81KBD
#ifdef ARDUINO_AVR_MEGA2560
const byte zx81pins[] = {37, 35, 33, 31, 29, 27, 25, 23, 47, 45, 43, 41, 39};
#else
const char zx81pins[] = {7, 8, 9, 10, 11, 12, A0, A1, 2, 3, 4, 5, 6 };
#endif
#endif

in hardware-arduino.h that does it. Never use the LED pin as a keyboard pin. 
The LED circuitry interferes with the debouncing on some boards.

twoporylyj, 2022-12-28:
A really very reconfigurable interoreter. So after i SET output to display and input to keyboard, the "echo" of pressed keys will be also sent to display?

slviajero, 2022-12-28:
Yes.

twoporylyj, 2022-12-28:
Therefore a standalone modern DIY Elektronika MK-85 like device, but 8-bit instead of 16, and with EEPROM instead of battery backup static RAM, is possible.

slviajero, 2022-12-28:
The interpreter can do 16 or 32 bit as well. Defining the address_t type to long will give an address space of 32 bit. I tried it on a Mac. If you have enough RAM like e.g. on an ESP32 or RP2040 you can build a 32bit BASIC computer. The memory access itself will be mapped to 8bit of course but all variables and arithmetic can be any type. You could also set number_t to a 64 bit type and the interpreter would adapt to this as well. 

> Am 28.12.2022 um 18:13 schrieb twoporylyj ***@***.***>:
> 
> MK-85

twoporylyj, 2022-12-28:
Compiler is even cooler, a similar feature exists in full version of Quick Basic (not Qbasic).

twoporylyj, 2022-12-28:
And also, when keyboard is not connected, and terminal is used, can i get "echo" of characters typed on terminal also sent to HD44780?

slviajero, 2022-12-28:
Yes. You can set the input stream to 1 i.e. the input characters of the terminal and the output stream to 2 i.e. the display. Then any typed character on the computer is echoed to the display and the computer acts as a keyboard.

twoporylyj, 2022-12-28:
10 @i=1:@o=2:input a
20 print a
run
32768
When i run this program, line 10 starts asking for number, and i type 32768 on terminal, characters appear on terminal, not echoed to HD44780. Then i press enter, line 10 prints 32768 on HD44780.

slviajero, 2022-12-28:
> 10 @i ***@***.*** <https://github.com/o>=2:input a

Do you use the picoserial option?

twoporylyj, 2022-12-28:
Yes, because without it i cannot type anything in the emulator i use (real hardware is not assembled yet because of lack of free time; i already have a nano and a display). Without picoserial enabled, the virtual terminal in emulator only outputs text, but not inputs it (on emulated nano).

slviajero, 2022-12-28:
Ok, that explains it. The Picoserial code cannot echo to a different device. This is because the picoserial is block oriented output and the echo needs to be done in an interrupt function. 

You can try to change this. Look for this function:

void picogetchar(char c){
	if (picob && (! picoa)) {
    picochar=c;
		if (picochar != '\n' && picochar != '\r' && picoi<picobsize-1) {
      if (c == 127 || c == 8) {
        if (picoi>1) picoi--; else return;
      } else 
			  picob[picoi++]=picochar;
      			picowrite(picochar);
		} else {
			picoa = 1;
			picob[picoi]=0;
			picob[0]=picoi;
			z.a=picoi;
			picoi=1;
		}
		picochar=0; /* every buffered byte is deleted */
	} else {
    	if (c != 10) picochar=c;
	}
}

Replace the picowrite(picochar) with outch(picochar). This may work on an LCD display but will certainly cause trouble on any slow display.
picogetchar is the interrupt handler. Going to outch() here is stretching the timing to the limit. On an nano it may work as there is not much interrupt business going on.

twoporylyj, 2022-12-28:
Thanks, using outch made this work as expected.

closed 2022-12-29 by slviajero

twoporylyj, 2022-12-30:
Even though issue is closed, thanks again. Tried this on real hardware, here is have i got it working:
1. Using real nano also requires activating picoserial, otherwise it does not react to typing text (so this is not bug in emulator). Outch also works same way as on emulator.
2. R/W line has to be grounded on real HD44780 unit i use; on emulator it does not matter.
3. Maximum contrast is when corresponding line of the display is fully grounded by the potentiometer, do not know whether all such displays behave like this one.
4. My display unit is 20x4, but i found that even this is configurable.
Keyboard is not built yet.
If being disconnected and then reconnected, nano (with CH340G) "jumps" either from /dev/ttyUSB0 to /dev/ttyUSB1, or back to /dev/ttyUSB0, depending on where it was previously. Baud rate is always 9600.

slviajero, 2022-12-30:
> Using real nano also requires activating picoserial, otherwise it does not react to typing text (so this is not bug in emulator). Outch also works same way as on emulator.

This is very strange. I just tested an LCD with an UNO and the hardware settings 

#undef USEPICOSERIAL
#define DISPLAYCANSCROLL
#define LCDSHIELD

Once I get the command prompt I use

SET 3,1 

to set the default output stream to the display. 

I then type on serial (using Coolterm and raw mode) and I get characters on the display. 

You need a terminal with raw mode (i.e. one that sends every character on keystroke) and you need to use SET3,1 or compile stream 2 as standard output stream. The behaviour you observe may be related with the terminal program you use.

> R/W line has to be grounded on real HD44780 unit i use; on emulator it does not matter.

Yes! They keep the display on Read mode as the good old LCD driver has no read feature. 
> Maximum contrast is when corresponding line of the display is fully grounded by the potentiometer, do not know whether all such displays behave like this one.
I think this is normal. 

> My display unit is 20x4, but i found that even this is configurable.

Exactly! It is the same driver for all geometries. 
> Keyboard is not built yet.
> If being disconnected and then reconnected, nano (with CH340G) "jumps" either from /dev/ttyUSB0 to /dev/ttyUSB1, or back to /dev/ttyUSB0, depending on where it was previously. Baud rate is always 9600.

I have see things like this as well.

twoporylyj, 2023-01-05:
I use minicom as terminal program, this may be the cause.