slviajero/tinybasic #46: Suggestion: extending the startup prompt

RonLinu, 2022-12-31:
Hi Stefan,
this is a suggestion (but I'm not quite sure this is the right forum to do it). 

Extending the startup prompt on the terminal to identify which version of TinyBASIC you are running.
There may be a smarter way to do it.
Located in "BASIC.H", starting at line 566:

#if defined(BASICINTEGER)
  const char mgreet[]		PROGMEM = "Stefan's Basic 1.4a (BASICINTEGER)";
#elif defined(BASICFULL)
  const char mgreet[]    PROGMEM = "Stefan's Basic 1.4a (BASICFULL)";
#elif defined(BASICSIMPLE)
  const char mgreet[]    PROGMEM = "Stefan's Basic 1.4a (BASICSIMPLE)";
#elif defined(BASICMINIMAL)
  const char mgreet[]    PROGMEM = "Stefan's Basic 1.4a (BASICMINIMAL)";
#elif defined(BASICTINYWITHFLOAT)
  const char mgreet[]    PROGMEM = "Stefan's Basic 1.4a (BASICTINYWITHFLOAT)";
#else
  const char mgreet[]    PROGMEM = "Stefan's Basic 1.4a";
#endif

Ron

RonLinu, 2022-12-31:
Incorrect suggestion!

closed 2022-12-31 by RonLinu

slviajero, 2023-01-01:
Hi Ron, why incorrect?