slviajero/tinybasic #47: String "addition"

twoporylyj, 2023-01-06:
Greetings. In some Basic interpreters the following is allowed:
10 a$="abc"
20 b$="def"
30 c$=a$+b$
40 print c$
abcdef
In this Basic attempt of doing this returns syntax error. It would be cool to implement this possibility here as well. Thanks.

slviajero, 2023-01-06:
The way to do it in Arduino IoT BASIC would be 

10 a$="abc“
20 b$="def"
30 c$=a$ : c$(4)=b$
40 print c$

I have adapted the substring logic of Apple Integer BASIC invented by Steve Wozniak.

You can use any substring as a lefthand or right hand value. Example

a$="abcdef“
b$="12"

print a$(2)
bcdef

print a$(1,2)
ab

a$(5)=b$
print a$
abcd12

a$="abcdef"
a$(1,2)=b$
print a$
12cdef

You always insert the string on the righthandside to the lefthandside 

There are a few examples how it is done here: https://github.com/slviajero/tinybasic/tree/main/examples/00tutorial <https://github.com/slviajero/tinybasic/tree/main/examples/00tutorial> (string1-string4)
and here: https://github.com/slviajero/tinybasic/tree/main/examples/08strings <https://github.com/slviajero/tinybasic/tree/main/examples/08strings>

twoporylyj, 2023-01-06:
Thanks, my cool new pushbutton-operated dual channel PWM dimmer with bargraphs now works.

10 set 3,1
20 for i=10 to 11
30 pinm i,1
40 next i
50 for i=14 to 17
60 pinm i,2
70 next i
80 a=0
90 b=0
100 if a>0 if dread(14)=0 a=a-1
110 if a<10 if dread(15)=0 a=a+1
120 if b>0 if dread(16)=0 b=b-1
130 if b<10 if dread(17)=0 b=b+1
140 c$="[-----------]  ":for i=0 to 10
150 if a>=i c$(i+2,i+2)="|"
160 next i:c$(14)=str(a)
170 d$="[-----------]  ":for i=0 to 10
180 if b>=i d$(i+2,i+2)="|"
190 next i:d$(14)=str(b)
200 cls:print c$
210 print d$;
220 awrite 10,a*25:awrite 11,b*25
230 delay 100
240 goto 100