slviajero/tinybasic #48: Question: Would it be possible to include "Adafruit_Floppy" Lib?

leandrofriedrich, 2023-01-17:
Hello, i was wondering if it would be possible to add https://github.com/adafruit/Adafruit_Floppy to this project to (for example) store basic programs on a floppy?

Greetings from Germany and i hope you are having a nice day so far!

slviajero, 2023-01-18:
I thought about it but for lack of hardware I can’t do this. Actually it would be very exotic and ver retro. 
Implementing it would not be hard. BASIC has a generic filesystem driver interface. You would have to implement the following functions 

void fsbegin(char);
int fsstat(char);
void filewrite(char);
char fileread();
char ifileopen(const char*);
void ifileclose();
char ofileopen(char*, const char*);
void ofileclose();
int fileavailable();

void rootopen();
int rootnextfile();
int rootisfile();
const char* rootfilename();
int rootfilesize();
void rootfileclose();
void rootclose();
void removefile(char )
void formatdisk(short);

The upper half is for opening and closing files. They would be enough for simple file operation.
Only if you want CATALOG, DELETE and FDISK, you would have to implement the second half of the functions.

leandrofriedrich, 2023-01-18:
thanks for the answer! ill dig out my pi pico over the next days and see if i can get something to work

slviajero, 2023-01-19:
I wrote a little tutorial on file system implementation:

https://github.com/slviajero/tinybasic/wiki/Peripherals:-SD-Cards,-SPIFFS,-littleFS,-EEPROM,-and-IO-channels.#build-the-file-access-functions <https://github.com/slviajero/tinybasic/wiki/Peripherals:-SD-Cards,-SPIFFS,-littleFS,-EEPROM,-and-IO-channels.#build-the-file-access-functions>

Maybe this helps. Best Regards, Stefan

slviajero, 2024-01-26:
Not in focus as hardware not really relevant.

closed 2024-01-26 by slviajero