slviajero/tinybasic #49: REM lines get distorted while loading source

guidol70, 2023-03-20:
Dear Stefan,
I tested IoT-BASIC on a TTGO VGA32. Its impressive fast :)
While it was PI-Day [some guys and I tested a programm which calcs Pi](https://forum.classic-computing.de/forum/index.php?thread/22627-runcpm-speed-vergleich-auf-verschiedenen-plattformen/&postID=406000#post406000).
It does need only around 3 secs for 100 positions after the comma!

In the ASCII-File source we got some REM lines at the begining and in the middle of the source:
[CPINEW.zip](https://github.com/slviajero/tinybasic/files/11021201/CPINEW.zip)

After loading the source-file the text after the REM gets distorted :(

Original sample:
```
100 REM CALCULATING PI FOR HUNDREDS OF DIGITS
110 REM THX TO ROSETTACODE.ORG FOR THE BASE OF THE PROGRAM
120 REM SHOULD BE A BENCHMARK FOR THE BASIC INTERPRETER/COMPILER
130 REM WRITTEN IN 2023 FOR FORUM.CLASSIC-COMPUTING.DE BY PETER DASSOW
```

after loading the source:
```
100 REM CALCULATINGPIFOR HUNDREDSOFDIGITS
110 REM THX TO ROSET TACODE0 OR GFOR THEBASEOFTHEPROGRAM
120 REM SHOULDBEABENCHMARKFOR THEBASICINT ERPRETER/COMPILER
130 REM WRITTENIN2023 FOR FOR UM0CLASSIC-COMPUT ING0 DEBYPETERDASSOW
```

Maybe you could check if the IoT-BASIC is trying to parse these REM-lines?

BTW: Es waere schoen, wenn Arrays einen Start-Indoex von 0 haetten.
         Ansonsten fehlt mir zur Zeit nur die Funktion CHR$
         und evtl. LIST Befehle wie "LIST -100" bzw. "LIST 100-500"

Liebe Gruesse nach Muechen und vielen Dank im Voraus fuers ueberpruefen im Voraus :)
Guido

slviajero, 2023-03-20:
IoT BASIC parses everything brutally and without exception.

The correct syntax of REM is 

10 REM “This is a comment”

Thank you for the compliments on “fast”. One reason is the tokenisation and then the line number cache on GOTO plus a few other tricks. I will try the PI program myself if I may.

Arrays can have a start index of zero by using a SET command. 

SET 12, 0 

Sets the array variable start index. Any number >= 0 is legal. The setting applies for all arrays. It can be changed any time.

LIST 100, 200 

Lists lines between 100 and 200.

LIST 0,100

shows all lines from beginning to 100. I use the same argument parser everywhere. That’s why it is not -100 or 100-200. 
The tokenised would make this a negative number expression. Brutal tokenisation has a price.

As to CHR$() this also exists in a slightly different form. Try

A$=“Hello”
A$(5)=65
PRINT A$

All string expression accept a number. Strings in IoT BASIC are like C strings: char arrays. 

If you want to print an ASCII directly, do 

PUT 65

instead of print. 

This works also the other way around. Try 

A$=“A”
PRINT A$
PRINT (A$)

;-). This is ASC(). I had a lot of fun doing this.

slviajero, 2023-03-22:
Hab übrigens das PI program gerade ausprobiert. Bin beeindruckt. Darf ich das als Beispiel in mein Tutorial mit aufnehmen? Beste Grüße, Stefan

guidol70, 2023-03-22:
> Hab übrigens das PI program gerade ausprobiert. Bin beeindruckt. Darf ich das als Beispiel in mein Tutorial mit aufnehmen? Beste Grüße, Stefan

Das duerfte kein Problem sein, wenn Du die Zeilen 100-130 mit drin laesst (inkl. den ""):

```
100 "REM "CALCULATING PI FOR HUNDREDS OF DIGITS"
110 "REM THX TO ROSETTACODE.ORG FOR THE BASE OF THE PROGRAM"
120 "REM SHOULD BE A BENCHMARK FOR THE BASIC INTERPRETER/COMPILER"
130 "REM WRITTEN IN 2023 FOR FORUM.CLASSIC-COMPUTING.DE BY PETER DASSOW"
```

Peter Dassow von http://www.z80.eu/ hatte die Version von https://rosettacode.org/wiki/Pi#BASIC
fuer das https://forum.classic-computing.de/forum/index.php angepasst.

Mit ihm zusammen habe ich dann fuer Dein IoTBasic die Anpassung fuer den Index-Anfang auf A(1) anstatt A(0) gemacht (bei IoTBasic startet der Index enzgegen der meisten anderen Basic bei 1)

Zum Schluss hatte ich noch die Zeit-Messung auf den Millis-Befehl angepasst und die RIGHT$ Funktion ersetzt mit denen aus IoTBasic ;)

Die Berechnung fuer Pi kam kurz nach dem Pi-Tag (daher die Idee). Normalerweise hatten wir im Forumsthread
https://forum.classic-computing.de/forum/index.php?thread/22627-runcpm-speed-vergleich-auf-verschiedenen-plattformen/&postID=403176#post403176
ein einfaches Mandelbrot in ASCII gezeichnet zum "Speed-Vergleich" von Z80/CPM-Systemen.
Aber es sind auch einige Real-Nur-BASIC-Systeme im Vergleich dabei :)

guidol70, 2023-03-26:
The new version of basic.c ([commit](https://github.com/slviajero/tinybasic/commit/24b3ffcc6f4a08f94539a9136f54de0f1d1ba08f) which support REMs without "" does work = tested, 
but [needs to initialize the varibales lexliteral & outliteral](https://github.com/slviajero/tinybasic/issues/52)

Looks fine after loading a .BAS:
```
130 REM "WRITTEN IN 2023 FOR FORUM.CLASSIC-COMPUTING.DE BY PETER DASSOW"
135 REM WRITTEN IN 2023 FOR FORUM.CLASSIC-COMPUTING.DE BY PETER DASSOW
```

slviajero, 2023-03-26:
Great! basic.h was old in the repo. Updated this. Need to test whether there are no side effects.

slviajero, 2023-03-27:
Fixed in 1.4. Literals now possible in REM.

closed 2023-03-27 by slviajero