slviajero/tinybasic #50: compiling BASIC on Windows 10 with MinGW/TDM-GCC

guidol70, 2023-03-20:
I did try to compile basic.c with
gcc -m32 basic.c -o basic-win.exe
on **MinGW32 (GCC 12.2.0)**
but do get the follwing errors (**and no .EXE**):
```
gcc -m32 basic.c -o basic-win.exe
In file included from basic.c:259:
hardware-posix.h: In function 'rootisfile':
hardware-posix.h:366:15: error: 'struct dirent' has no member named 'd_type'
  366 |   return (file->d_type == DT_REG);
      |               ^~
hardware-posix.h:366:27: error: 'DT_REG' undeclared (first use in this function)
  366 |   return (file->d_type == DT_REG);
      |                           ^~~~~~
hardware-posix.h:366:27: note: each undeclared identifier is reported only once for each function it appears in
```

When Iam using **TDM-GCC (GCC 10.3.0)** I do get the follwing errors/warnings (**BUT do get a .EXE**):

```
gcc -m32 basic.c -o basic-win.exe
In file included from basic.c:259:
hardware-posix.h:55:8: warning: 'timeb' is deprecated [-Wdeprecated-declarations]
   55 | struct timeb start_time;
      |        ^~~~~
In file included from basic.h:51,
                 from basic.c:245:
C:/TDM-GCC-32/include/sys/timeb.h:57:32: note: declared here
   57 | struct __POSIX_2001_DEPRECATED timeb
      |                                ^~~~~
In file included from basic.c:259:
hardware-posix.h: In function 'timeinit':
hardware-posix.h:56:1: warning: 'ftime' is deprecated [-Wdeprecated-declarations]
   56 | void timeinit() { ftime(&start_time); }
      | ^~~~
In file included from basic.h:51,
                 from basic.c:245:
C:/TDM-GCC-32/include/sys/timeb.h:169:6: note: declared here
  169 | void ftime (struct timeb *__t) {_ftime ((struct _timeb *)(__t)); }
      |      ^~~~~
In file included from basic.c:259:
hardware-posix.h: In function 'rtcget':
hardware-posix.h:161:9: warning: 'timeb' is deprecated [-Wdeprecated-declarations]
  161 |  struct timeb thetime;
      |         ^~~~~
In file included from basic.h:51,
                 from basic.c:245:
C:/TDM-GCC-32/include/sys/timeb.h:57:32: note: declared here
   57 | struct __POSIX_2001_DEPRECATED timeb
      |                                ^~~~~
In file included from basic.c:259:
hardware-posix.h:163:2: warning: 'ftime' is deprecated [-Wdeprecated-declarations]
  163 |  ftime(&thetime);
      |  ^~~~~
In file included from basic.h:51,
                 from basic.c:245:
C:/TDM-GCC-32/include/sys/timeb.h:169:6: note: declared here
  169 | void ftime (struct timeb *__t) {_ftime ((struct _timeb *)(__t)); }
      |      ^~~~~
In file included from basic.c:259:
hardware-posix.h: In function 'millis':
hardware-posix.h:258:9: warning: 'timeb' is deprecated [-Wdeprecated-declarations]
  258 |  struct timeb thetime;
      |         ^~~~~
In file included from basic.h:51,
                 from basic.c:245:
C:/TDM-GCC-32/include/sys/timeb.h:57:32: note: declared here
   57 | struct __POSIX_2001_DEPRECATED timeb
      |                                ^~~~~
In file included from basic.c:259:
hardware-posix.h:259:2: warning: 'ftime' is deprecated [-Wdeprecated-declarations]
  259 |  ftime(&thetime);
      |  ^~~~~
In file included from basic.h:51,
                 from basic.c:245:
C:/TDM-GCC-32/include/sys/timeb.h:169:6: note: declared here
  169 | void ftime (struct timeb *__t) {_ftime ((struct _timeb *)(__t)); }
```

I do get a prompt from v1.4b, but the first thing I did try was CLS and I only do get a char which doesnt clear the screen:

[image]

I do get this char in Cmder or in the normal Windows-console. 
**In both CLS does work with the v1.31 from your Release Page.**

How do you compile the Windows version and with which version of a MinGW?

Kind regards
Guido

slviajero, 2023-03-21:
Currently I compile the Windows version with a very old MINGW, gcc 6.3.3. The version 10 gives the error messages because the old deprecated time API is used. This is fairly harmless but not good.

I am worried about the gcc12 result. I will look into this. On Mac gcc 12 with all the libraries works well. There is probably some header / include path issue. 

As to the terminal sequences: they are not implemented in Windows and the POSIX versions. You can send ANSI standard sequences through the PUT command. I have never tested or implemented this. Will have a look at it also.

slviajero, 2023-03-21:
Just for illustration on my Mac with ANSI sequences activated in the terminal 

PUT 27, "[", "2", "J"

is 

CLS

If you want to make that permament in BASIC, the lines to change would be 

			case TCLS:
				xc=od; 
/* if we have a display it is the default for CLS */
#if defined(DISPLAYDRIVER) || defined(GRAPHDISPLAYDRIVER)		
				od=ODSP;
#endif
				outch(12);
				od=xc;
				nexttoken();
				break;

Instead of the outch(12) you would need the sequence above. This would be the quick and dirty way of doing it. I will add a better way later for ANSI ready terminals.

slviajero, 2023-03-22:
> Did that "dirty" way with 4 outch-lines - worked well ;)
> Where did you get "MINGW, gcc 6.3.3"?
> I only did find a installer <https://sourceforge.net/projects/mingw/files/latest/download> for 6.0.3, but this didnt complete the installation :(

Hi Guido, 

I got a clean implementation uploaded now in the repo. You need the new hardware-posix.h.

Why clean? Escape characters should be handled in the hardware specific file. I never implemented ANSI and the VT52 sequences in the Windows / POSIC code but will do this now including the pseudo VT52 terminal.

The MINGW I have is from 2016. It is a 6.3.0 gcc. I do not develop much on Windows, mostly Mac and the various microcontrollers. I will set up a fresh MINGW environment on the weekend and then look at these things.

Best, 
Stefan

EvilDaveCanada, 2023-03-22:
Just noticed your thread.

One suggestion I have is to research the UNIX/Linux Termcap and Terminfo
terminal definitions. I know there are Termcap entries out there for use
with MSDOS and possibly the old command prompts of Win 3.x, Win9.x and so
on. There may even be one for the newer Windows command prompts such as cmd
& the Linux Sub-System command prompts and scripts. AWK is another place to
look.

I used to program in a Business Basic called BBx by Basis. They now have a
version that will compile a BBx program written in the BBx interpreter in
to pure Jave script. I'm retired now but I know that there are a few
attempts to create a BBx look-a-like on GitHub. They all seemed to get the
commands working including commands for Displaying on a Display. What no
one seems to know what to do is replicate the BBx file structures. It's too
bad that I never knew about those attempts as I have detailed notes on how
they would work as well as my own replacement file designs. I just was
never a low level programmer so I could only design the system but had to
still use BBx or the newer version called BBj.

slviajero, 2023-03-22:
> Hi Stefan,
> I reverted the "dirty" way in basic.c and used your new clean hardware-posix.h and it worked right away :)
> I always like the clean way more thany "hacking" in a drity way.
> Many thanks!
> PS: I checked the Manual and missed something like a way to quit (other than Ctrl-C) IoTBASIC.
> On other BASIC-flavours there is a command like system/bye/quit to leave the interpreter via a command.

CALL 0

;-), Will add this to the manual.

guidol70, 2023-03-23:
> Hi Guido,  
I will set up a fresh MINGW environment on the weekend and then look at these things. 

Hi Stefan,
after rereading the compile error from MinGW32 (GCC 12.2.0) I did found out (as I did read on the net) that the dirent.h downst have the **d_type** support anymore.

So - I dont know if it is a "BIG HACK - but while comparing the dirent.h from GCC 10.3.0 and 12.2.0 I did see only 2 lines missing for **d_type** in the defined structure 
```
struct dirent
  unsigned        d_type;		/* File attributes */
struct _wdirent
  unsigned        d_type;		/* File attributes */
```
and a missing section for 
**if defined(_BSD_SOURCE) || defined(_WIN32)** regarding the DT_REG & d_type enabled 1
```
#define DT_REG   	_A_NORMAL
#define _DIRENT_HAVE_D_TYPE    1
```

So I added these to the dirent.h file of GCC 12.2.0 and could compile the BASIC without errors :)
But the .EXE is now around 81KB bigger (110KB<->191KB)

Kind regards
Guido

[dirent_h_ORIGINAL.TXT](https://github.com/slviajero/tinybasic/files/11050861/dirent_h_ORIGINAL.TXT)
[dirent_h_PATCHED.TXT](https://github.com/slviajero/tinybasic/files/11050816/dirent_h_PATCHED.TXT)
[basic-win-binarys.zip](https://github.com/slviajero/tinybasic/files/11050855/basic-win-binarys.zip)

slviajero, 2023-03-23:
Hi Guido, 

can you tell me which MinGW you use? 

I compile against the old, only 32, „official“, MINGW coming from https://sourceforge.net/projects/mingw/ <https://sourceforge.net/projects/mingw/>
This was the first one. 

Then there is the new one https://www.mingw-w64.org <https://www.mingw-w64.org/> with binaries from https://sourceforge.net/projects/mingw-w64/ <https://sourceforge.net/projects/mingw-w64/> 

They use different headers and APIs. 

Best Regards,
Stefan

guidol70, 2023-03-23:
I think I do use the MSVCRT-Runtime version of this MinGW32 (GCC-12.2.0 Win32):
https://winlibs.com/#download-release

Not the UCRT-Version, because UCRT is only in newer Windows-Version available

slviajero, 2023-03-23:
Got it and understood it. Try the new version of basic.c and hardware-posix.h I just checked in. Should compile and run without problems. Please set the MINGW64 instead of MINGW in basic.c.

Info: dirent in mingw64 is different. Walking through a dir has to be done differently, less like Unix, more like Windows. I switched off the file type test (just like in MSDOS) for now.

slviajero, 2023-03-27:
Fixed in 1.4 code. Added MINGW64 flag. Further investigation needed on how to handle directory on Windows systems.

closed 2023-03-27 by slviajero