slviajero/tinybasic #51: LOAD command doenst clear the memory before loading

guidol70, 2023-03-25:
Hi Stefan,
I dont know if it is intended to work so, but when I got a program in memory and do a LOAD the loaded program will be merged with the program in memory 
(did see this when having a program without REMs in memory and loading CPINEW2.BAS).

MANUAL.md doesnt state that we have to use NEW before loading a new program or that the programs will be merged.
On other systems the loaded program will completly replace the program in memory.

PS: the BASIC does also compile with the [MinGW32 GCC 13.0.1](https://github.com/brechtsanders/winlibs_mingw/releases/download/13.0.1-snapshot20230122-10.0.0-msvcrt-r1/winlibs-i686-posix-dwarf-gcc-13.0.1-snapshot20230122-mingw-w64msvcrt-10.0.0-r1.7z) Snapshot from WinLibs ;)

Kind regards
Guido

slviajero, 2023-03-25:
;-). Thank you! Added this to the manual. Actually in interachtive mode LOAD = MERGE while in run mode LOAD = CHAIN i.e. truly replaces the program but not the data.

slviajero, 2023-03-27:
Documented feature in the manual.

closed 2023-03-27 by slviajero