slviajero/tinybasic #52: compile problem in Linux armbian (debian on ARM-CPU)

guidol70, 2023-03-26:
didnt got this 
**undeclared (first use in this function)**
error in debian on X86, but today on armbian while I did try to compile:

```
basic.c: In function ‘nexttoken’:
basic.c:2398:13: error: ‘lexliteral’ undeclared (first use in this function)
 2398 |         if (lexliteral) {
basic.c:2398:13: note: each undeclared identifier is reported only once for each function it appears in
basic.c: In function ‘outputtoken’:
basic.c:4843:34: error: ‘outliteral’ undeclared (first use in this function)
 4843 |         if (token == LINENUMBER) outliteral=0;

```

solved that with a init of lexliteral & outliteral in this part of the code (before int main):
```
/* debug mode switch */
#define DEBUG 0

/* Init lexliteral & outliteral */
int lexliteral;
int outliteral;

/*
 * the core basic language headers including some Arduino device stuff
 */
#include "basic.h"
```

**BTW:** with cpinew*.bas & Linux the output of the digits is **printed after the complete calculation** and not while calculation :(  On Windows and TTGO VGA32 it seems calcualting is faster than printing on the screen, because it looks like the measured time is shorter than it took to print the numbers on the screen

slviajero, 2023-03-26:
My apologies, I forgot to check in basic.h - fixed now. 

> Am 26.03.2023 um 15:31 schrieb Guido Lehwalder ***@***.***>:
> 
> 
> didnt got this
> undeclared (first use in this function)
> error in debian on X86, but today on armbian while I did try to compile:
> 
> basic.c: In function ‘nexttoken’:
> basic.c:2398:13: error: ‘lexliteral’ undeclared (first use in this function)
>  2398 |         if (lexliteral) {
> basic.c:2398:13: note: each undeclared identifier is reported only once for each function it appears in
> basic.c: In function ‘outputtoken’:
> basic.c:4843:34: error: ‘outliteral’ undeclared (first use in this function)
>  4843 |         if (token == LINENUMBER) outliteral=0;
> BTW: with cpinew*.bas & Linux the output of the digits is printed after the complete calculation and not while calculation :( On Windows and TTGO VGA32 it seems calcualting is faster than printing on the screen, because it looks like the measured time is shorter than it took to print the numbers on the screen
> 
> 
Yes! I noticed this as well.

slviajero, 2023-03-27:
basic.h now hopefully fixed, thank you for bringing it up ;-)

closed 2023-03-27 by slviajero