slviajero/tinybasic #54: Feature request - Basic embedded in MycroPython

Tangerino, 2023-05-18:
One interesting use case for the BASIC interpreter is to be integrated as part of Micropython project.

A subset of the language is enough, exemple: No I/O.
So an end user can upload its own custom code to an IOT device and perform custom actions.

This is safe, fast, small and not expose any hardware capability.

Imagine a control system, where the input is a temperature sensor and output is a 1 or 0 to be sent to an actuator.

What is inside the BASIC function is the customer property.

slviajero, 2023-05-22:
Interesting idea. But how should this look like in practice? I am not familiar with the MicroPython project. My interpreter can be integrated into any environment by creating a hardware-xxx.h file. This file needs to implement the necessary I/O functions . It also needs a thread that calls setup() once and then loop() permanently. So, generally speaking, everything is prepared to embed it in a function.

slviajero, 2024-01-26:
Closed, could not be specified more accurately.

closed 2024-01-26 by slviajero