slviajero/tinybasic #56: SH110x OLED support

Rowan-Bird, 2023-07-21:
I have an SH1106G OLED, which is incompatible with the SSD1306 library. I can't seem to find any include lines to modify

slviajero, 2023-07-21:
> I have an SH1106G OLED, which is incompatible with the SSD1306 library. I can't seem to find any include lines to modify
> 
> 

Integrating a new display is a bit tricky. The display driver code is very complex because it needs to handle many different displays. 

First you need to add a macro identifying your display, for example ARDUINOSH1106G. Something like 

#define ARDUINOSHG1106H 

should be set.

Includes and definitions are handled around code lines 560 to 940 of hardware-arduino.h Best copy the Nokia51 code. Look for the ARDUINONOKIA51 macro and copy this code. I assume that you connect the Oled to SPI. This should be very similar to the Nokia. 

Then look at code lines 1215-1263. This is how a display driver is connected to a physical display. You need to implement all the functions in this section with the library you are using. Again, the NOKIA51 is a good example. 

If you need help, please contact me. This is one of the more difficult projects with the BASIC interpreter.

Rowan-Bird, 2023-07-22:
Mine is connected via I2C. Anyway, would I be able to use U8X2? (I'm using an Arduino UNO so RAM is a small issue)
Also, with my 16x2 LCD, I can't get a PS/2 keyboard to work, I'm using an IBM SK-8840 and have verified that it works on a computer. 

I don't have a spare PS/2 socket but I do have copious amounts of these,
[image]
however I can't get anything to happen. I did enable ARDUINOPS2 and STANDALONE, anything else I did wrong/should be?

slviajero, 2023-07-23:
> Am 22.07.2023 um 05:48 schrieb RowanCrE ***@***.***>:
> 
> 
> Mine is connected via I2C. Anyway, would I be able to use U8X2? (I'm using an Arduino UNO so RAM is a small issue)
> Also, with my 16x2 LCD, I can't get a PS/2 keyboard to work, I'm using an IBM SK-8840 and have verified that it works on a computer.
> 
> 

For the PS2 keyboard, which pins did you use? The default pins in the code are 2 and 5. If you used the LCDSHIELD macro to connect a 16x2 this would use the standard pin definitions with pins 4, 5, 6, 7 as data pins which would cause a pin conflict. You could use pin 3 for the PS2 data line.

If this is not the problem, there are two other things you could try: 
1. Memory: Try to reduce the BASIC memory by setting MEMSIZE by hand to a lower value, like e.g. 512. This leaves more space for buffers etc. I don’t think this is the problem, but at least it is an option.
2. Power: a PC2 keyboard needs a lot of power. Sometimes they don’t initialize properly at startup. An external 2A power supply of even a separate power supply for the keyboard can help.

As to the I2C display with u8x8: this could be integrated just like an ARDUINOLCDI2C as a text only display. I will have a look at this myself next week, when I am on vacation. Never thought about the text only option for I2C oleds. 

Best,
Stefan

slviajero, 2023-07-23:
I checked the code. There was a bug with PS2 and keypad handling. The keypad caused the PS2 code to read wrong data. This is now fixed with a workaround in TinybasicArduino. Please download this code and try it out. Memory doesn’t seem to be a problem, but I needed and external power supply for the keyboard to start properly. Still, the software bug was the main reason why it didn’t work.

Rowan-Bird, 2023-07-23:
Alright, I'll try again. My 16x2 LCD is I2C btw, but I have a parallel one as well.

slviajero, 2023-07-23:
Ok, the I2C should have worked always. You can use the ARDUINOLCDI2C macro and set the right parameters (rows columns). The bug only affected the parallel ones when used with the LCDSHIELD macro. This macro activates the keypad of the shield as input device. And this overrides the PS2 in the old code.

slviajero, 2024-01-26:
Seems to be closed. No further feedback.

closed 2024-01-26 by slviajero