slviajero/tinybasic #60: A couple of questions about Keyboards

souphead, 2023-10-25:
I have a Lilygo TTGO and the Basic2 IoTBasic seems to run. But I have a couple of questions.

First I have US keyboard so I went into runtime.cpp to change it from the German one. No problem, easy fix. But are you planning to take the keyboard definitions out of runtime.cpp and move the selection workings into hardware.h like the rest of the hardware definitions?

Secondly, the backspace moves the cursor back a position bit it doesn't seem like it is doing a delete and the delete key doesn't appear to be doing anything, are those the actions you expect? (maybe the German keyboard works differently)

slviajero, 2023-10-29:
> Am 25.10.2023 um 03:03 schrieb souphead ***@***.***>:
> 
> 
> I have a Lilygo TTGO and the Basic2 IoTBasic seems to run. But I have a couple of questions.
> 
> First I have US keyboard so I went into runtime.cpp to change it from the German one. No problem, easy fix. But are you planning to take the keyboard definitions out of runtime.cpp and move the selection workings into hardware.h like the rest of the hardware definitions?
> 
> 

Must go there! I forgot this. Thank you for mentioning this. 

> Secondly, the backspace moves the cursor back a position bit it doesn't seem like it is doing a delete and the delete key doesn't appear to be doing anything, are those the actions you expect? (maybe the German keyboard works differently)
> 
> 

That seems to be an error with the keyboard translation. It worked in BASIC V1. Normally the ASCII characters 8 and 127 are  both interpreted as delete. This should be independent of the keyboard type.

Can you do me a favour and try to find out which ASCII character your keyboard produces. Try 

10 GET C
20 IF C<>0 THEN PRINT C
30 DELAY 100
40 GOTO 10 

in BASIC and tell me the codes. This is tricky because FabGL has a builtin VT52 emulation. Can be that this changes something. Can also depend on the version.

Best, 
Stefan

slviajero, 2023-10-30:
Checked the code now.

What happens is: Backspace actually does delete and goes back but does not display it on screen. Will fix this.
There is another bug. GRAPH is switched off on FabGL. 

Will fix both now.

> Am 29.10.2023 um 06:47 schrieb Stefan Lenz ***@***.***>:
> 
> 
> 
>> Am 25.10.2023 um 03:03 schrieb souphead ***@***.*** ***@***.***>>:
>> 
>> 
>> I have a Lilygo TTGO and the Basic2 IoTBasic seems to run. But I have a couple of questions.
>> 
>> First I have US keyboard so I went into runtime.cpp to change it from the German one. No problem, easy fix. But are you planning to take the keyboard definitions out of runtime.cpp and move the selection workings into hardware.h like the rest of the hardware definitions?
>> 
>> 
> 
> Must go there! I forgot this. Thank you for mentioning this. 
> 
>> Secondly, the backspace moves the cursor back a position bit it doesn't seem like it is doing a delete and the delete key doesn't appear to be doing anything, are those the actions you expect? (maybe the German keyboard works differently)
>> 
>> 
> 
> That seems to be an error with the keyboard translation. It worked in BASIC V1. Normally the ASCII characters 8 and 127 are  both interpreted as delete. This should be independent of the keyboard type.
> 
> Can you do me a favour and try to find out which ASCII character your keyboard produces. Try 
> 
> 10 GET C
> 20 IF C<>0 THEN PRINT C
> 30 DELAY 100
> 40 GOTO 10 
> 
> in BASIC and tell me the codes. This is tricky because FabGL has a builtin VT52 emulation. Can be that this changes something. Can also depend on the version.
> 
> Best, 
> Stefan

souphead, 2023-10-31:
Sorry my internet was down. 
So I ran your program and I print 7 for Delete and 8 for Backspace. Hope that helps.

souphead, 2023-10-31:
I pulled the the Oct 30th copy of the code. I made only two changes to hardware.h - **#define TTGOVGA** and **#define ARDUINOKBDLANG US** (and commenting out the German entry). Seems it is picking up the German keyboard.

Am I missing another setting in hardware.h I need to make?

If I pick TTGOVGA, I don't see how ARDUINOUSBKBD, PS2KEYBOARD or PS2FABLIB are getting set. Having said that it seems that PS2FABLIB is defined because ARDUINO_TTGO_T7_V14_Mini32 is defined somehow.

[image]

I guess I'm confused!

slviajero, 2023-11-01:
fixed it now. I start to confuse languages and used a wrong syntax for C preprocessor statements. 
should work now. 

I also looked in to the backspace thing. Actually this comes from the interpretation of the 
backspace character in the VT52 emulation of FabLib. Sending a 8 or a 127 to the terminal 
in both cases moves the cursor one position back but does not delete the character. 
This is probably the correct behaviour for a VT52 terminal. 

One could fix it by sending a VT52 sequence back to the terminal to delete the character. 
I hesitate to put this in the code right now. Need to think about terminal emulation a bit more.

souphead, 2023-11-01:
I can confirm the US keyboard works now, thanks!

As to the issue of the delete action, it can be a whole load of work depending on how many features of editing you want to support. A project like Kilo Text Editor - https://viewsourcecode.org/snaptoken/kilo/ handled all that stuff with a VT100 terminal emulation but there is a lot of complexity there.

slviajero, 2023-11-02:
Actually, I have build a complete VT52 emulation which is already part of BASIC. All hardware platforms I support use this emulation. You need this once you want to build stand alone systems with displays and keyboards. Even the tiny 16*2 LCD can be controlled with VT52 sequences in BASIC. On Mac and Linux I wrote a VT100 to VT52 converter to keep the interpreter compatible. In language.h you find the HASVT52 directive that is almost always on. It controls the compile of this part of the interpreter. Only for Fablib, I bypass my own VT52 and use Fablibs components.

Why VT52? Because it is the smallest command set that can do full screen text editing like for Kilo. Sequences are mostly just one byte. I can be implemented easily. On an Arduino UNO, the VT52 component needs 1kB of flash.

I need to think about the future of the FabLib port. Where to put a thin compatibility layer for the build in VT52 and the emulation of the interpreter. It could be placed in inch or consins but that doesn’t feel right.

slviajero, 2024-01-26:
Fixed.

closed 2024-01-26 by slviajero