slviajero/tinybasic #61: Running code from micro sd cards?

vrdriver, 2023-12-10:
Hi, I love this concept of running basic on an Arduino.
Is there presently any way of loading programs that are stored on an sd card and running them that way?
Have I misread the concept of this project, or does it all have to by typed in?
Sorry to be a bother
Steve.

slviajero, 2023-12-10:
sure! in hardware.h set the ARDUINOSD macro. only good if you have enough memory. most boards like the nano every, mega, r4 are ok.Sent from my iPhone

vrdriver, 2023-12-11:
Hey thanks! I'll take a look at it. I have a UNO and a MEGA, so maybe I'll be able to get things running on the MEGA.

slviajero, 2023-12-11:
Yes! There are a few tricks. If you run in trouble, tell me. 
Some SD cards have odd timing problems.

slviajero, 2024-01-26:
Closed, already implemented.

closed 2024-01-26 by slviajero