slviajero/tinybasic #62: error: void value not ignored as it ought to be

Suresh-Subedi, 2024-02-26:
I am trying to build this basic for arduino uno using platformio. I copied all .ino code to main.cpp. I copied other files in the same src folder with same names. I also added #include <Arduino.h> to all the files. When I compile I encounter following two errors and compilation fails. What could be causing this?

src\main.cpp: In function 'number_t getvar(mem_t, mem_t)':
src\main.cpp:1100:30: warning: invalid conversion from 'mem_t (*)(address_t) {aka signed char (*)(unsigned int)}' to 'mem_t {aka signed char}' [-fpermissive]       
  return getnumber(a, memread2);
                              ^
In file included from src\main.cpp:37:0:
src\basic.h:437:6: note:   initializing argument 2 of 'void getnumber(address_t, mem_t)'
 void getnumber(address_t, mem_t);
      ^~~~~~~~~
### src\main.cpp:1100:30: error: void value not ignored as it ought to be
  return getnumber(a, memread2);
                              ^
src\main.cpp: In function 'void setvar(mem_t, mem_t, number_t)':
src\main.cpp:1167:27: warning: invalid conversion from 'void (*)(address_t, mem_t) {aka void (*)(unsigned int, signed char)}' to 'mem_t {aka signed char}' [-fpermissive]
  setnumber(a, memwrite2, v);
                           ^
### src\main.cpp:1167:27: error: too many arguments to function 'void setnumber(address_t, mem_t)'
In file included from src\main.cpp:37:0:

slviajero, 2024-02-26:
This looks like the header and the interpreter file don't fit together in the repo. I changed the entire getnumber and setnumber logic. Looks like I checked in an inconsistent state. I will check this tonight.

slviajero, 2024-02-26:
Please try to use basic.h from the Posix folder in the meantime. This should be ok.

Suresh-Subedi, 2024-02-26:
If I chage return value to char and add address_t to signature in header, I get:

src\main.cpp:2326:5: error: ambiguating new declaration of 'int innumber(number_t*, char*, address_t)'
 int innumber(number_t *r, char* buffer, address_t k) {
     ^~~~~~~~
In file included from src\main.cpp:37:0:
src\basic.h:517:6: note: old declaration 'char innumber(number_t*, char*, address_t)'
 char innumber(number_t*,char*,address_t);
      ^~~~~~~~
src\main.cpp: In function 'void outnumber(number_t)':

slviajero, 2024-02-26:
> I am gettings another error when I use basic.h from Posix folder. I also copied basic.c as it doesn't exist in IotBasic.
> src\basic.c: At top level:
> src\basic.c:2322:5: error: conflicting types for 'innumber'
> In file included from src\basic.c:35:0:
> src\basic.h:517:6: note: previous declaration of 'innumber' was here
> src\basic.c: In function 'parsestringvar':

Ok, let me check tonight. I will check in a consistent state. In my local dev environment it works cool.

slviajero, 2024-02-26:
I fixed this now. basic.h and IotBasic.ino are consistent again. Sorry for the mistake. 
I change a lot in the types and data structure right now in the 2.0alpha version. In about 2 weeks 
2.0 will got beta and be stable again.

Suresh-Subedi, 2024-02-27:
Thank you. Unfortunately I still have some errors:

### src\main.cpp:1273:30: error: void value not ignored as it ought to be
  return getnumber(a, memread2);
                              ^
src\main.cpp: In function 'void setvar(mem_t, mem_t, number_t)':
src\main.cpp:1348:27: warning: invalid conversion from 'void (*)(address_t, mem_t) {aka void (*)(unsigned int, signed char)}' to 'mem_t {aka signed char}' [-fpermissive]
  setnumber(a, memwrite2, v);
                           ^
### src\main.cpp:1348:27: error: too many arguments to function 'void setnumber(address_t, mem_t)'
In file included from src\main.cpp:35:0:
src\basic.h:436:6: note: declared here
 void setnumber(address_t, mem_t);
      ^~~~~~~~~
src\main.cpp: In function 'address_t createarray(mem_t, mem_t, address_t, address_t)':

Suresh-Subedi, 2024-02-27:
I am using basic.h from IotBasic again. Should I still use basic.h from Posix?

slviajero, 2024-02-27:
is identicalSent from my iPhone

slviajero, 2024-02-27:
but best use both files from IoTBasic. I have tested this file set.Sent from my iPhone

Suresh-Subedi, 2024-02-28:
I cloned your repo. Downloaded and installed Arduino IDE and compiled repo in it. It compiled fine. PlatformIO is a bit more quirky and requires methods to be declared before they are used.

> src\main.cpp:1348:27: error: too many arguments to function 'void setnumber(address_t, mem_t)'
In file included from src\main.cpp:35:0:
src\basic.h:436:6: note: declared here
void setnumber(address_t, mem_t);
^~~~~~~~~
src\main.cpp: In function 'address_t createarray(mem_t, mem_t, address_t, address_t)':

I solved this error by placing `void setnumber(address_t m, memwriter_t f, number_t v);` after imports in main.cpp (IotBasic.ino).

> src\main.cpp:1348:27: error: too many arguments to function 'void setnumber(address_t, mem_t)'
In file included from src\main.cpp:35:0:
src\basic.h:436:6: note: declared here
void setnumber(address_t, mem_t);
^~~~~~~~~
src\main.cpp: In function 'address_t createarray(mem_t, mem_t, address_t, address_t)':

While investigating this, I saw that method signature of `getnumber` doesn't match in basic.h (`void getnumber(address_t, mem_t);`) and IotBasic.ino (`number_t getnumber(address_t m, memreader_t f) {...}`). I got further by updating getnumber in basic.h to return number_t. Still encountering other issues though when linking: collect2.exe: error: ld returned 1 exit status.

slviajero, 2024-02-28:
Ok! Absolutely correct. These were still the wrong prototypes. My apologies.
I now checked in the corrected version. 

Background info: I changed the entire mem access code and I am still rewriting. I always test first on the Posix code. So this is a remnant of this process.

slviajero, 2025-06-22:
Closed and fixed.

closed 2025-06-22 by slviajero