slviajero/tinybasic #64: Arduino GigaR1 build settings

EncomLab, 2024-05-14:
What are the build settings to be used for the Arduino GigaR1?

slviajero, 2024-05-19:
I haven’t got a Giga right now, so I haven’t tried. 
Did you try it? Any error messages etc? 
The interpreter should compile right through but EEPROM and other peripherals might be tricky. 
Also, the memory heuristics will be wrong. Best would be to set MEMSIZE explicitly. 
Best, Stefan

easyprototype, 2025-02-04:
Hi. I am also interested in the build settings for the GigaR1.

I tried the following for BASIC 2.0:
in hardware.h
#define EEPROMSIZE 4096

in language.h
#define BASICFULL

I got the following compilation error:
`In file included from C:\Projets\Arduino\ArduinoGIGA\IoTBasic\runtime.cpp:15:0:`
`runtime.h:739:13: error: conflicting declaration 'typedef int PinStatus'`
` typedef int PinStatus;`
`             ^~~~~~~~~`

This error repeats itself over many files.

Thanks :)

slviajero, 2025-02-04:
Can you just uncomment the typedef and see what happens?
Some boards bring this typedef with them and some don’t. This is because the development environments of various Arduinos are not really compatible.

easyprototype, 2025-02-04:
uncomment?

I'm don't understand. Do you mean to put this line in comment so it is no longer compiled?

slviajero, 2025-02-04:
yes sorry, comment it so that it is inactive Sent from my iPhone

easyprototype, 2025-02-04:
Ok, that works! The project has compiled, I have uploaded it into my ArduinoGigaR1 and I am talking to it using CoolTerm! I don't get the welcome message, though :(

The program is 167k (8% of available flash) and uses 49k (9% of available RAM) of RAM.

Now I need to test the I/O and Serial devices. At the moment I am trying to know how much memory BASIC sees but I don't know how. "print himem" returns 127.

One question though: what am I to do with EEPROMSIZE? The Giga doesn't have any EEPROM, just plenty of flash. Do I just define EEPROMSIZE with however space I want to save my programs?

Thanks.

slviajero, 2025-02-04:
Something is wrong here, if print himem says 127 then probably at start the GIGA will have returned 128 as max BASIC memory. This is because memory allocation needs a heuristic and this needs to be programmed. 

Best set memory size manually. This is done with

#define MEMSIZE 32000

for a 32kB system in hardware.h.

EEPROM is a different story. Let me take a look and then answer this later.

easyprototype, 2025-02-04:
"print himem" now returns 31999. I also uploaded and executed successfully "battle.bas". That seems to work :)

You know your code very well, thank you very much!

I can write 1s and 0s to digital 2, but writing anything to A0 eludes me...
The "led" label is also functioning. I can turn on and off the user LED.
The "delay" command works.

slviajero, 2025-02-07:
> On 4. Feb 2025, at 20:54, easyprototype ***@***.***> wrote:
> 
> 
> "print himem" now returns 31999. I also uploaded and executed successfully "battle.bas". That seems to work :)
> 
> You know your code very well, thank you very much!
> 
> I can write 1s and 0s to digital 2, but writing anything to A0 eludes me...
> The "led" label is also functioning. I can turn on and off the user LED.
> The "delay" command works.
> 
> 

That looks promising. A0 is difficult. You need to access the analog lines with there native i.e. digital pin numbers. The AZERO macro is not reliable. 
A0 should be pin 76. The command to write it would be AWRITE 76,512 and to read would be AREAD(76). I have not tested this. Will get my Giga board today and then check on it.

The Giga has no EEPROM emulation. The 16 GB flash can be accessed as a block device but there are no simple libraries. So your EEPROM 4096 setting will not work as this would use an EEPROM emulation class just like on the ESP. In the absence of this class the command is ignored. I look into this this weekend or maybe next weekend. During winter time my programming activity strongly depends on the skiing weather in the Alps. The last to days were splendid so I was out there.

easyprototype, 2025-02-07:
Yes, I read that A0 and up are labeled 76+. the AREAD() function works up to A7. From A8 and up it does not seems to function.

PINM, DREAD & DWRITE works for A0 to A7, but not for A8+.

Any AWRITE on pin 76 crashes big time the ArduinoGiga. The red LED starts blinking. I can tell you the exact same thing happens when I perform a analogWrite(76, xx) in the Arduino IDE. So it is not a problem in the BASIC kernel.

I enabled and tried I2C on stream &7. Looking at my scope, that seems to work :)

For the EEPROM problem, thanks for checking it.

My next test is enabling ARDUINOSD and connect a SDCard connector directly on the GIGA, no shield involved. Which pins must/should I use? I must use SPI5 (because SPI1 is 5V). After scanning the IotBasic files, I see nowhere I can specify the Pins for the SD card.

Although, I could use the USB A port with a USB flash drive instead, if that is possible with the BASIC kernel.

Have a nice skiing time :)

slviajero, 2025-02-07:
> Yes, I read that A0 and up are labeled 76+. the AREAD() function works up to A7. From A8 and up it does not seems to function.
>
> PINM, DREAD & DWRITE works for A0 to A7, but not for A8+.
>
> Any AWRITE [...]

Same here. Tried it with my board. I didn’t look into it very deeply. Looks like the board uses A8-A11 as „pure analog pins“ which seems to be something special. I also assume that AWRITE only works on some (digital) pins. On D2 it works fine. 

> I enabled and tried I2C on stream &7. Looking at my scope, that seems to work :)
> 
Good!
> For the EEPROM problem, thanks for checking it.
>
> My next test is enabling ARDUINOSD and connect a SDCard connector directly on the GIGA, no shield involved. Which pins must/should I use? I must use [...]

This is really complicated. First of all, I tried SPI on the central header and it works fine in BASIC with a standard card adapter. I use the ones that can do 3.3 and 5 V. I followed this recipe: 

https://forum.arduino.cc/t/arduino-giga-r1-and-sd-card/1221185/5 <https://forum.arduino.cc/t/arduino-giga-r1-and-sd-card/1221185/5>

With the pinout in the middle of this thread BASIC finds the SD card without any additional settings.

SPI parameters cannot be configured right now. They are hardcoded in spibegin() in runtime.cpp. You can set them there.
BASIC only knows one SPI object. 

Unfortunately things are even more complicated. The SD library uses the SPI object. This is hardcoded there. 
The Giga board configures SPI as the central one and the other one is SPI1. Even if you start SPI1 in spibegin()
the SD library knows nothing about it. 

One can patch the SD library. I have done this in the past for some use cases. But then you have two different SD
libraries with different *.h files. If you want this, I can set it up, but I would not recommend it. The central SPI works fine. 

I also created a clean code for the startup and memory heuristic. It is checked in already. This code starts the 
interpreter with 64kB memory and correct settings. 

> Although, I could use the USB A port with a USB flash drive instead, if that is possible with the BASIC kernel.
> 
I looked at this and will play with it tomorrow. 
> Have a nice skiing time :
> 
> 

Thank you! A bad weather front came over the Alps. Am grounded for a few days.

slviajero, 2025-02-07:
By the way, the names in the documentation 

https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet/ <https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet/>

are wrong. In the variants file there are different definitions

SPI is the center port with D89-D91. The cheat sheet says it is SPI1 but this is not true.

SPI1 is D11, D12, D13. The cheat sheet says it is SPI5 but this is also not true. 

I used D10 as a CS for the SD card.

slviajero, 2025-02-07:
PS: further down in the documentation they correct themselves and mention the difference in schematics and code. 
In my opinion both are 3.3V. I doubt that the side pins are 5V safe. 

> On 7. Feb 2025, at 22:52, Stefan Lenz ***@***.***> wrote:
> 
> By the way, the names in the documentation 
> 
> https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet/ <https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet/>
> 
> are wrong. In the variants file there are different definitions
> 
> SPI is the center port with D89-D91. The cheat sheet says it is SPI1 but this is not true.
> 
> SPI1 is D11, D12, D13. The cheat sheet says it is SPI5 but this is also not true. 
> 
> I used D10 as a CS for the SD card. 
> 
>

slviajero, 2025-02-08:
Checked the voltage tolerance 

https://forum.arduino.cc/t/arduino-giga-r1-wifi-input-pin-voltage-tolerant/1301148 <https://forum.arduino.cc/t/arduino-giga-r1-wifi-input-pin-voltage-tolerant/1301148>

all cool on both SPI ports and the I2C all of them 5V and 3.3V. 

I2C display works.

easyprototype, 2025-02-08:
> I tried SPI on the central header and it works fine in BASIC with a standard card adapter.

On my side I checked the ArduinoGiga schematics. All SPI lines comes directly unbuffered from the STM32. I will confirm with my scope first on monday but I will use the same SPI port as you. That will simplify things greatly, no more coding involved. Thanks!

> I also created a clean code for the startup and memory heuristic. It is checked in already. This code starts the
interpreter with 64kB memory and correct settings.

Thank you for updating tinyBasic for the Giga, I really appreciate.

>> Although, I could use the USB A port with a USB flash drive instead, if that is possible with the BASIC kernel.

> I looked at this and will play with it tomorrow.

That would be grand if it works :)

My next todo after I set up the SD card will be to look how to implement new BASIC commands. I remember I saw somewhere (in the code I believe) that we can do so. Let's face it, I am not using a Giga just for the speed. I want to use the hardware peripherals. I would like to implement a PEEKREG16, POKEREG16, PEEKREG32 and POKEREG32 command in order to access the STM32 registers, which are not 8-bit and not mapped inside BASIC memory. Naming is to be determined, I can go up to 16 characters by default, right?

slviajero, 2025-02-08:
> On my side I checked the ArduinoGiga schematics. All SPI lines comes directly unbuffered from the STM32. I will confirm with my scope first on monday but I will use the same SPI port as you. That will simplify things greatly, no more coding involved. Thanks!
> 
> 

I will have a look at the SPI bus again later. But so far it works. I also added a 32 kB EEPROM now on the I2C. This also works and BASIC says this at startup:

> Stefan's Basic 2.0 Memory 65535 32767
> print @e
8191

> My next todo after I set up the SD card will be to look how to implement new BASIC commands. I remember I saw somewhere (in the code I believe) that we can do so. Let's face it, I am not using a [...]

The lexer code is a bit weird. So here is how it works. Everything is done in nexttoken(). A keyword is isolated and it can be only upper case or lower case letters, @ and _. No numbers allowed. This is then checked against the keyword table. If found the lexer finds the token and returns it. 

Then it is interpreted as a variable and (only) there numbers are also allowed. This is inconsistent and was done to make the interpreter work in very small systems. Don’t ask me why.

The length of the keyword buffer is controlled by SBUFSIZE and this is 32 by default. The length of the variable name string is controlled by MAXNAME and this is also 32. 

If you want to implement an access to low level registers, there would be many ways. The simplest is to extend the USR command and the SET command. They are generic commands for this. Minimal coding effort. Then you can also create a new command as you proposed. But no numbers in commands. The last method would be to use a special variable of the form @<letter>. There is a stub for @U which could be extended easily. I wrote a tutorial about this:

https://github.com/slviajero/tinybasic/wiki/How-to-extend-BASIC <https://github.com/slviajero/tinybasic/wiki/How-to-extend-BASIC>

Best,
Stefan

slviajero, 2025-02-08:
The GIGA USB device works now. 

#define GIGAUSBFS 

activates the code. I tested it very superficially. Looked good. 

The build in real rime clock of the GIGA also works using 

#define ARDUINORTC

I am impressed. Both were easy to implement because they used a good POSIX API.

The documentation on the GIGA website of Arduino is not that good.

easyprototype, 2025-02-08:
You are going way too fast for me! I can only work on it on monday!

> I also added a 32 kB EEPROM now on the I2C. This also works and BASIC says this at startup:
>> Stefan's Basic 2.0 Memory 65535 32767
 >>print @e
8191

Do you mean you added an external EEPROM chip on the I2C bus, or are you using some EEPROM/flash available on the Giga?

Thank you for pointing me out your tutorial on extending tinyBASIC. I will read it carefully.

>The GIGA USB device works now.

I can't wait to try this. If I enable GIGASBFS, I must not enable/use SD Cards, right?

>The build in real rime clock of the GIGA also works

Oh, the joy!

>The documentation on the GIGA website of Arduino is not that good.

Indeed, that's why I use the STM32H747 datasheets + the GIGA schematics.

Now that I think of it, I have a question: is there a BASIC command that lists all the commands available in the build? In Forth, typing "WORDS" display on screen all the words the system knows and that you can use.

slviajero, 2025-02-08:
> Do you mean you added an external EEPROM chip on the I2C bus, or are you using some EEPROM/flash available on the Giga?
> 
External. That is clean code that I wrote a while ago. It uses no library except one of my own. 
> I can't wait to try this. If I enable GIGASBFS, I must not enable/use SD Cards, right?
> 
Right! If ARDUINOSD is on, all other filesystems are set to off automatically. 
So, only #define GIGAUSBFS and no other filesystem driver. 

> Indeed, that's why I use the STM32H747 datasheets + the GIGA schematics.
> 
> Now that I think of it, I have a question: is there a BASIC command that lists all the commands available in the build? In Forth, typing "WORDS" display on screen all the words the system knows and that you can use.
> 
> 

No, never implemented an online help. This takes a lot of flash. On the GIGA it would be doable, though. The source for all keywords would be the manual right now.

easyprototype, 2025-02-09:
Not an Online help, just a dump of the keyword strings from the string storage (what is listed at the beginning of ioTBasic.ino). Basically a parser that prints all the tokens instead of finding the one you just typed.

slviajero, 2025-02-09:
Done. I introduced the HELP command for this. In its current form it does just what you proposed. I will work in it a bit later. Quick and dirty implementation right now.

slviajero, 2025-02-10:
I just took a look at this. 

> Compilation returned
> runtime.cpp:5055:10: fatal error: SoftwareSerial.h: No such file or directory
> #include <SoftwareSerial.h>
> 
> So I added a " !defined(ARDUINO_ARCH_MBED_GIGA)" to the IF before it.
> 
> 

Right! I added that to the main code as well only that I went further and made it a !defined(ARDUINO_ARCH_MBED) because all the MBEDs don’t have SoftwareSerial
> 
> Everything compiled, downloaded and booted up. Sent stuff on the &4 and &7 stream. Works fine. If I try memory size
> 
> print himem
> 65534
> 
> I typed HELP. Beautiful! That's exactly that. From here, you know what your system supports or not. With that you go read the online manual. No need to implement it in the program. Brilliant :)
> 
> 

Great!
> Next I tried using the real time clock. I am unfamiliar on how to set time. I tried writing directly at @t <https://github.com/t>(x) to set time but the I always read back
> 
> print @t <https://github.com/t>$
> 45:85:85-45/25/65
> 
> So I undef ARDUINOWIRE and ARDUINOPRT but kept ARDUINORTC.
> 
> Now when I type
> 
> print @t <https://github.com/t>$
> 
> Basic hangs up, the red LED starts blinking and I have to reset.
> 
> Don't know what else to do because of my lack of RTC knowledge so moving on.
> 
> 

So, really, this is misleading in the code. If you #undef ARDUINORTC it uses the build in RTC. Then @T$ and all the rest works. If you #define ARDUINORTC then a serial I2C RTC is expected and as you have none the Wire library blocks. This is, by the way, a bug in Wire that was reported a million times and never fixed. 
> Disabled RTC, enabled GIGAUSBFS. Compilation succeeded after I installed the USBHostMbed5 library. Basic now takes significantly longer to boot (show prompt on console after reset).
> 
> CATALOG, LOAD and SAVE works, but I must first boot the GIGA with my USB key inserted. That is fantastic. Is there a way to mount/unmount the key in the system without rebooting?
> 
> 

I checked this and played around with this. I managed to shorten the startup time a bit but I did not manage to get the msd or usb device accept a plugged in USB stick after start of BASIC. There is a trick here but it doesn’t seem to work.

slviajero, 2025-02-10:
> On 10. Feb 2025, at 18:03, easyprototype ***@***.***> wrote:
> 
> 
> Just tried enabling ARDUINOSD.
> 
> On the scope the lines on the SPI port are 3.3V so I connected my SD card adapter directly on it, with D10 as CS and of course GND + 3.3V as the power supply.
> 
> 

Good! 
> Like the USB key, I must boot the Giga with the card inserted. CATALOG, LOAD and SAVE works super fine. Thanks!
> 
> 

One can restart the filesystem driver by using CALL 1 in BASIC. CALL 0 restarts the system. 

CALL 1 was intended to make SD card change possible but it somehow doesn’t always work. I never analysed this. 

With USB it works never. I check this right now.

easyprototype, 2025-02-10:
Thanks for the heads up.

With ARDUINOPRT, ARDUINOWIRE and ARDUINOSD enabled (+ARDUINORTC disabled) I get

>print @t$
14:16:36-2/1/72

When I set @t(0/1/2/4), the Giga seems to count time.

@t(6) seems to be the year instead of weekday.
@t(5) seems to be the day of the month instead of the year.
@t(3) is unknown.

I tried Call 1 with an SD card and indeed it works.

slviajero, 2025-02-10:
The definitions are 

@T(0): seconds (0-59)
@T(1): minutes (0-59)
@T(2): hours (0-23) (24 hour clock mode only is supported)
@T(3): day of week with range 0-6
@T(4): date - day (1-31)
@T(5): date - month (1-12)
@T(6): date - year (0-99)

One has to set each one individually by assigning to @t(x). I have not found time to implement a better way. 
t(3) may be broken.

slviajero, 2025-02-10:
PS: just played around with the GIGA USB keyboard library. Unfortunately really not so good. They say it is alpha and it is.

easyprototype, 2025-02-10:
>@t0): seconds (0-59)
@t(1): minutes (0-59)
@t(2): hours (0-23) (24 hour clock mode only is supported)
@t(3): day of week with range 0-6
@t(4): date - day (1-31)
@t(5): date - month (1-12)
@t(6): date - year (0-99)

Ok. It's just that MANUAL.MD lists @t(4) to @t(6) differently.

slviajero, 2025-02-11:
Oh, that is indeed bad. Question is now, is the code inconsistent or the manual wrong. Will check right now.

easyprototype, 2025-02-11:
Thanks for clearing up the RTC variables.

Also, I noticed that load/saving files on USB uses the RTC for setting the file date/time. Super nice.

Now I must make design choices.

1) I would like to compile in BASICFULL (float), but I need to handle unsigned 32bit integers. So I must use BASICINTEGER which is 32 bit signed, but I can live with that. Is it possible to handle 32bits integer with TinyBasic compiled in floating point?

2) In order to access registers, I think I would make 2 commands: RDREG addr,size and WRREG addr,size,value. All arguments are 32bits and WRREG needs 3 arguments. Can I create commands with 3 arguments?

3) Is it possible to write and display values in hexadecimal in TinyBasic?

Thanks!

slviajero, 2025-02-11:
> I would like to compile in BASICFULL (float), but I need to handle unsigned 32bit integers. So I must use BASICINTEGER which is 32 bit signed, but I can live with that. Is it possible to handle 32bits integer with TinyBasic compiled in floating point?
> 
> 
Not really. I never implemented a second number type. I mostly worked with 8bit situations, where it doesn’t matter. Boolean logic is done by casting floats to 16bit integers, than AND/OR/NOT them and then they are cast back to float. The MAP command, however, uses 32 bit internally. 

How exactly would you want to use them?

> In order to access registers, I think I would make 2 commands: RDREG addr,size and WRREG addr,size,value. All arguments are 32bits and WRREG needs 3 arguments. Can I create commands with 3 arguments?
> 
> 
Is the reader a command or a function? Would you return a value or use it more in the way INPUT is done? 

If it is a function, this would mean that the syntax would be 

X=RDREG(addr, size)

and for WRREG it would be 

WRREG addr, size, X

If it is a INPUT type command, the syntax would be 

RDREG addr, size, X 

and the result of the read would go into X. 

Both is possible and there are examples in BASIC. 

A two argument function would be USR. From this you can learn how it is done. 

There are two elements that need to be implemented. First, within factor, there needs to be something like this: 

    case TUSR:
        parsefunction(xusr, 2);
        break;

Parsefunction is a helper that collects n numerical arguments. In this case n=2.

Then you need to define a helper function that looks like this 

void xusr() {
    address_t fn;
    int arg;

    v=pop();
    fn=pop();

I deleted a few lines. The arguments are on the stack and can simply be retrieved with pop(). All this including the error handling is done with parsefunction. Simply follow the pattern of USR and you get a result.

Important: XUSR does not call nexttoken() at the end. Functions called from factor() do not do this. 

If you want to create a command, WIRE would be a good example how to do it. 

Again, there are two elements. You need something in statement() that looks like this: 

        case TWIRE:
            xwire();
            break;

And then you need a helper function that looks like this: 
void xwire() {
    short port, data1, data2;

    nexttoken();
#if defined(HASWIRE) || defined(HASSIMPLEWIRE)
    parsearguments();
    if (!USELONGJUMP && er) return; 

    if (args == 3) {
        data2=pop();
        data1=pop();
        port=pop(); 
        wirewriteword(port, data1, data2);
    } else if (args == 2) {
        data1=pop();
        port=pop(); 
        wirewritebyte(port, data1);
    } else {
        error(EARGS);
        return;
    }
#endif
}

So, how does it work? First, and really important, nexttoken() needs to be called. Why is that so? Because the program cursor in statement() is pointing to the command token. This is TWIRE. The variable token has the value TWIRE. 

nexttoken() is an iterator. In a modern language an iterator construct would be used. In C you do this by hand. If nexttoken() is called, the cursor is advanced to whatever token comes after WIRE. 

If you just want to read three numerical arguments, use the helper parsearguments(). This function processes a comma separated list of numerical arguments. VERY VERY IMPORTANT: this function handles all the errors and then returns with the er variable set. the „if“ after parsearguments() must be there because if „er" is set, something went wrong and it makes no sense to continue. 

After this you have the number of arguments in args and the arguments on the stack from where you can pop(). If an error occurs somewhere in your code, simply call error() with one of the allowed arguments and then return from the function. The call stack of functions handles all errors upstream and either branches to interactive mode or calls an error handler. 

Two more thoughts: 
1. if you want to read from a function and write with a command, you can use the same keyword, for example REGISTER

REGISTER addr, size, value

and 

X=REGISTER(addr, size) 

are different from the point of view of the interpreter. One is a function in statement() and the other is a function in factor(). 

If you chose to do this, i’d propose  either REGISTER or PORT as a single keyword. 

2. You can do the register handling in the .ino file. But of course it would be cleaner to do the hardware dependent part as a function in runtime.cpp. Best follow the pattern in WIRE.
> Is it possible to write and display values in hexadecimal in TinyBasic?
> 
> 
No, I have not yet included functions for this. Also no format for hex constants. Must be done in the program.

slviajero, 2025-02-11:
> easyprototype
>  left a comment
> (slviajero/tinybasic#64)
> If it is a function, this would mean that the syntax would be
> X=RDREG(addr, size)
> and for WRREG it would be
> WRREG addr, size, X
>
> I would make both of them functions, it mimics DWRITE, DREAD and PINM.
>
> I like the idea of using the same keyword for reading and writing, but I think it will be confusing for me when I look at my program.
>
> You gave me plenty of meat to digest. If I manage to make them work, I will also try to add two new operators: >> and << (logical shift)...

If you follow the patterns of PINM DWRITE and DREAD nothing can go wrong! Good luck with it. 

With the operators, this is slightly harder as you need to modify expression(). Once you get there, I can give you a few tips.

easyprototype, 2025-02-19:
Hi again.

Following your tutorial, I managed to implement one function and one command. With the HELP command, I even see them appear with the other commands :)

I need to thoroughly test my read command first, I get weird readings depending of the address and the data size. I need to read a little about memory addressing on the STM32, like can I read 2 bytes (16 bits) when starting from an odd address?

Next, I would like to implement a decimal to hexadecimal and a hexadecimal to decimal command. Hexadecimal numbers would be text strings though. So I need to read a little on how strings are passed around in TinyBasic.

Which brings me to a question/request:

1) can you add the available I/O streams to the HELP command?

2) In doing so, you would update TinyBasic, forcing me to re-implement my commands into the new TinyBasic version. I am trying to find a way to do this easily. For example, I can make two .h files. One is my declarations and is included at the beginning of IoTBasic.ino like any .h file. The second is included somewhere appropriate in the middle of IoTBasic.ino and contains all my functions. Not clean but it would save me a lot of cut and paste. After that I still need to manually add stuff the the lists, statement() and factor() functions. Any idea?

Thanks,

Serge

slviajero, 2025-02-19:
> Following your tutorial, I managed to implement one function and one command. With the HELP command, I even see them appear with the other commands :)

Cool!
> I need to thoroughly test my read command first, I get weird readings depending of the address and the data size. I need to read a little about memory addressing on the STM32, like can I read 2 bytes (16 bits) when starting from an odd address?
> 

This can be a nightmare. Usually the compiler does this for you in one way or the other and often it is not portable. In the BASIC main code I try to use 8bit and then reconstruct 32 bit values from a 8 bit dataflow. This is slow but safe and portable.
If you only program one platform that is native 32bit, I probably would use uint32_t as data type and then explicitly construct 16 bit upper and lower word by shifting and masking.

> Next, I would like to implement a decimal to hexadecimal and a hexadecimal to decimal command. Hexadecimal numbers would be text strings though. So I need to read a little on how strings are passed around in TinyBasic.
> 
> Which brings me to a question/request:
> 
> can you add the available I/O streams to the HELP command?
> 
> 
Sure. I thought to do this anyway. Will look at it.
> In doing so, you would update TinyBasic, forcing me to re-implement my commands into the new TinyBasic version. I am trying to find a way to do this easily. For example, I can make two .h files. One [...]
I was thinking of splitting BASIC into components and subsystem in the near future but haven’t made up my mind yet. It will take about 20 hours of work to split BASIC into modules and test it.
But of course, extending BASIC and then still having to upgrade to newer versions is tedious. 

Let me think about this a little.

easyprototype, 2025-02-20:
> In the BASIC main code I try to use 8bit and then reconstruct 32 bit values from a 8 bit dataflow.

I try not to reconstruct because of how processors store 16 and 32 bits values. On the STM32 the low address is the LSB, while on a x86 processor the low address is the MSB. So I leave it to the compiler. I declare a uint8_t pointer for reading 8 bit values, a uint16_t pointer for 16 bit values and so on. Anyway I also crashed my Giga just by reading a memory address that maybe I shouldn't have so I need to make further testing. I will also try to test it on my Mega to see if my function works on a different processor.

>I was thinking of splitting BASIC into components and subsystem in the near future but haven’t made up my mind yet. It will take about 20 hours of work to split BASIC into modules and test it.

Don't do that. I made myself a walkthrough that I added as comments in my functions' file. I can re-implement my functions in a new version of TinyBasic in about 10 minutes. Much faster than 20 hours!

slviajero, 2025-02-20:
Code can now display a banner with the IO channels …

easyprototype, 2025-02-20:
Ooooh, nice banner. Very informative.

I got to try my re-implementation procedure: less than 10 minutes and my Giga was up and running again :) I like that you added the available IO in your banner. However I don't see stream #4 (ARDUINOPRT) which I enabled in my build. Also, what is stream 0 exactly?

slviajero, 2025-02-20:
Oh, little bug. Fixed now.

easyprototype, 2025-02-21:
Tried a new build this morning. All streams accounted for in the new banner :)

However, the banner reports

>Language set: full

even as I build using "#define BASICINTEGER". And indeed I can add floating numbers at the prompt. I find the comment in the header of language.h confusing. "NOLANGUAGEHEURISTICS" does not exist. However, by reading carefully the code, I see that by default, LANGUAGEHEURISTICS is defined and override all the following defines. So I "#undef LANGUAGEHEURISTICS" and now I am in integer mode.

slviajero, 2025-02-21:
I changed the name of the variable. Felt better like this.

slviajero, 2025-02-22:
> Me again.
> I am having trouble making my read memory address works. As far as I can see, I am having problems handling a true 32bit number with push() and pop(). I am compiling with BASICINTEGER.
> I want to read TIM14->CCR1. It contains 9000 and the decimal address is 1073750068. I never read the correct address. So I validated that my function rdreg received the correct address. So I tried to understand the various pop() and push() functions available and came up with
> address_t x,y;
> x=popaddress();
> y=popaddress2();
> pushaddress2(y);
> By using a address_t, popaddress(); and pushaddress2() (see previous), if I type
> print rdreg(1073750068, 4)
> 1073750068
> This is the only way I get back the exact first number typed as argument. Using pop() and push() only returns me the lowest 2 bytes of the number.
> Then I tried
> address_t x,y,result;
> // now we just ignore x and y
> // perform a brutal direct memory read
> result = *(uint32_t *)1073750068UL;
> pushaddress2(result);
> This works and returns me 9000.
> address_t y, result;
> y=1073750068UL;
> result = *(int32_t *)y;
> This also works. But if I do
> // use address typed at the BASIC prompt
> Then I get an incorrect value.
> How do I correctly pass my 32bit address to my C funtion?

I didn’t look at this very thoroughly, will have more time tomorrow, but maybe one observation

address_t is a type defined in basic.h. It is normally uint16_t, this means a unsigned 16 bit integer. popaddress()  is made to store this type safely on the stack. This means you are casting to a 16bit number. 

If you compile BASIC anyway as a integer basic, you can use number_t, push() and pop(). They handle 32bit objects because BASIC uses this type in the code. But of course it is a signed type.

In general, you can use any type you want for number_t and address_t with just one rule: number_t has to be at least as big as address_t. The interpreter should run with number_t being a a long, i.e. a 64 bit integer. With this number_t is so large, that you can use 32 bit unsigned numbers and store them in this type.

slviajero, 2025-02-23:
> 
> 
> easyprototype
>  left a comment 
> (slviajero/tinybasic#64)
> Thanks.
> 
> To me, it is not clear if the int data type is 32bits on a STM32.
> 
> I misread the size of address_t and at first, pop() and push() did not work, probably because I used address_t variables with them.
> 
> I will try again Monday using t_number variables, pop() and push().
> 
> 
> 
> easyprototype
>  left a comment 
> (slviajero/tinybasic#64)
> Thanks.
> 
> To me, it is not clear if the int data type is 32bits on a STM32.
> 
> I misread the size of address_t and at first, pop() and push() did not work, probably because I used address_t variables with them.
> 
> I will try again Monday using t_number variables, pop() and push().

This is tricky. I struggled with this when I first started to introduce address_t. The key to understand it is here in basic.h:

If BASIC is a 16 bit BASIC, means it has a 16 bit internal address space with maximum memory of 65536, then an unsigned 16 bit type is used for this address space. This is uint16_t.
address_t is used everywhere in the string, array and heap code. It is an internal pointer type of BASIC. If one increases it to a 32 bit type, all internal objects in BASIC get bigger and the interpreter is slower. The memory of the interpreter is aways a stream of 8 bit objects. This is mem_t. If address_t is 32 bit, 4 read or write operations are needed to store it. Storing and recalling of mem_t is the performance bottleneck. If you want an interpreter that can run on 8 bit and 32 bit CPUs, this is the only way of doing it. At least, I didn’t find another way. 

On an integer BASIC, number type is always int and int is 32 bit. I tested long int here which is 64 bit and it should work also. 

On a GIGA, slowing down of the interpreter or memory limits should be a concern. The machine is so powerful.

You could do one thing without changing the code. Try to set MEMSIZE explicitly to a value greater than 65536. Like for example #define MEMSIZE 128000. Then address_t is bigger automatically as you can see in the code. 

#ifdef HASFLOAT
typedef float number_t;
typedef long wnumber_t;
#else
typedef int number_t;
typedef int wnumber_t;
#endif
/* default is 16bit addresses and max 64k memory, setting MEMSIZE in hardware.h overrides this */
#if MEMSIZE == 0 && MEMSIZE < 65536
typedef uint16_t address_t; /* this type addresses memory */
#else
/* use this for large memory systems, tested occassionally */
typedef uint32_t address_t;
#endif
typedef int8_t mem_t; /* a signed 8 bit type for the memory */
typedef int index_t; /* this type counts at least 16 bit */
#ifndef HASLONGTOKENS
typedef int8_t token_t; /* the type of tokens, normally mem_t with a maximum of 127 commands and data types */
#else
typedef int16_t token_t; /* token type extension, allows an extra of 127 commands and symbols */
#endif

slviajero, 2025-02-25:
> 
> 
> easyprototype
>  left a comment 
> (slviajero/tinybasic#64)
> Arg, I was so close and I wasted so much time!
> 
> 
> 
> easyprototype
>  left a comment 
> (slviajero/tinybasic#64)
> Arg, I was so close and I wasted so much time!

Ask me, how much time I needed to sort the types out so that the code stayed portable. 
Multiplatform portability is a headache. 
> I also wiped out a quick and dirty direct write to register using only number_t variables and pop(). I can now change the duty cycle of my 16 bit PWMs on the fly :)
> 
> Thanks for the help. Next: accessing a string of text in a function.

Hint: look at the getstring() and stringvalue() function. Identifying a string in the token stream is done with stringvalue() which also 
does getstring().

The most important thing here is to know where the string is. Some strings are in the C-space of the memory. Example: the input buffer, 
the time buffer and some other special strings. Some strings are in the BASIC memory i.e. on the heap. 

Best read stringvalue() carefully. 

Please ask if you have any questions. This is difficult.

easyprototype, 2025-02-26:
After checking how strings are handled, I think, for making my function HEXTODEC(string), I will copy and modify factorval() and implement it the same way in factor().

My function requires a string containing a hexadecimal written in ASCII and converts it into a decimal number. This looks a lot like the behavior of the VAL command.

slviajero, 2025-02-28:
I just checked in a code prototype that makes VAL capable of processing strings like „0xFA“, „0o76“, „0b1001“ for hex, octal and binary data. Maybe you can use this? With this there would be no need for an additional command. The code is off by default and has to be activated in language.h with the flag HASNUMSYSTEM. I was thinking to add this also to STR.

Still largely untested. Sign handling is odd.

easyprototype, 2025-02-28:
>I just checked in a code prototype that makes VAL capable of processing strings like „0xFA“, „0o76“, „0b1001“ for hex, octal and binary data. Maybe you can use this?

Wow, I most certainly can, my next todo was implementing binary to decimal. It's very useful when working with registers.

>I was thinking to add this also to STR.

Oh wow, that would be cool, another of my todo list :)

If I can help you in any capacity, please let me know!

slviajero, 2025-02-28:
> 
> 
> easyprototype
>  left a comment 
> (slviajero/tinybasic#64)
> I just checked in a code prototype that makes VAL capable of processing strings like „0xFA“, „0o76“, „0b1001“ for hex, octal and binary data. Maybe you can use this?
> 
> Wow, I most certainly can, my next todo was implementing binary to decimal. It's very useful when working with registers.
> 
> I was thinking to add this also to STR.
> 
> Oh wow, that would be cool, another of my todo list :)
> 
> If I can help you in any capacity, please let me know!
> 
> 
> 
> easyprototype
>  left a comment 
> (slviajero/tinybasic#64)
> I just checked in a code prototype that makes VAL capable of processing strings like „0xFA“, „0o76“, „0b1001“ for hex, octal and binary data. Maybe you can use this?
> 
> Wow, I most certainly can, my next todo was implementing binary to decimal. It's very useful when working with registers.
> 
> I was thinking to add this also to STR.
> 
> Oh wow, that would be cool, another of my todo list :)
> 
> If I can help you in any capacity, please let me know!

Yes, actually you could help. Can you test the VAL() and the STR() command. As many test cases as possible. Syntax is currently 

A=VAL("0xFCB0“)  for hex
A=VAL("0b10011101“) for binary
A=VAL("0o7632“) for octal

The letters x,b,o are case insensitive. Maximum string length for conversion is 32 characters which may be a problem in binary.

Syntax for STR is 

A$=STR(number, base“)

STR should be able to covert to any base. It will output the result without the prefix 0x or 0o or 0b. 

If you write a small test program with the cases that are relevant for you I would use it then in my unit test library. 

Also, can you send me what you did for the port access?

slviajero, 2025-02-28:
I now added << and >> as an operator for bit shift. They cast number_t to the int of the platform and then shift.

Also added a function BIT which tests one bit. Use would be something like

IF BIT(A,5) THEN 

BIT returns a boolean false or true. This makes some things easier.

Tests and feedback would be highly appreciated.

easyprototype, 2025-03-01:
Ok, just compiled your new kernel with BASICINTEGER + my stuff and it boots.
Quick'n'dirty test of <<, >> and BIT. It works :)
Quick'n'dirty test of VAL and STR. It works :)

STR being able to convert into any numerical base just took me back to my elementary school years, when they tried to teach me that... Now I just converted 25 in base61 for fun.

I will try to write a test program for VAL and STR. Once it is done, I will run it on my GIGA in an integer kernel and then in a float kernel. After that I will try the same on an ArduinoMega.

How do I send you my code for port access?

P.S. In manual.md, when you explain the BIT command, you wrote
> If the bit is not set the answer is always 0. If the bit is not set the answer will be either -1 or 1.

I think there is a mistake is the wording. The bit is not set in both cases.

slviajero, 2025-03-01:
> Ok, just compiled your new kernel with BASICINTEGER + my stuff and it boots.
> Quick'n'dirty test of <<, >> and BIT. It works :)
> Quick'n'dirty test of VAL and STR. It works :)
> 
Good!
> STR being able to convert into any numerical base just took me back to my elementary school years, when they tried to teach me that... Now I just converted 25 in base61 for fun.
> 

That was also what I felt. Tested it with counting in the 3 system. I still remember my teacher who made me do this.

> I will try to write a test program for VAL and STR. Once it is done, I will run it on my GIGA in an integer kernel and then in a float kernel. After that I will try the same on an ArduinoMega.
> 
> How do I send you my code for port access?
> 
Mail me to ***@***.*** ***@***.***>. That is simpler than GITHUB.
> P.S. In manual.md, when you explain the BIT command, you wrote
> 
> If the bit is not set the answer is always 0. If the bit is not set the answer will be either -1 or 1.
> 
> I think there is a mistake is the wording. The bit is not set in both cases.
> 
> 

Corrected! Thank you.

I have programmed some quick and dirty code for port access on the UNO. It doesn’t use a command but instead has a special array @P(). @P(0) is PORT B and so forth. @P(16) … are the data direction registers and @P(32) … are the PIN registers for input read. 

Best,
Stefan

easyprototype, 2025-03-03:
Just sent you my files to your email address that I found in your github profile because the one you mentioned here got stripped by github :(

slviajero, 2025-03-05:
> I modified my direct register access commands so they now compile and work in float. Unfortunately for me, they don't work in float because any number bigger than 16777214 (24 bits) is transformed into scientific notation. And as we get closer to this limit the number becomes rounded.
> I tried this in BASICFULL on my Giga
> 3010 PRINT "Testing imprecision and biggest integer number before swapping to scientific notation"
> 3020 A=16777207: PRINT "16777207 =", A
> 3030 A=16777208: PRINT "16777208 =", A
> 3040 A=16777209: PRINT "16777209 =", A
> 3050 A=16777210: PRINT "16777210 =", A
> 3060 A=16777211: PRINT "16777211 =", A
> 3070 A=16777212: PRINT "16777212 =", A
> 3080 A=16777213: PRINT "16777213 =", A
> 3090 A=16777214: PRINT "16777214 =", A
> 3100 A=16777215: PRINT "16777215 =", A
> 3110 A=16777216: PRINT "16777216 =", A
> Results are interesting.

Is this because BASIC converts to scientific? This could be changed easily. I could add a number format modifier similar to what the old programmable pocket calculators had.

easyprototype, 2025-03-05:
>Is this because BASIC converts to scientific?

Yes, it's because of the way float numbers are stored. The bigger they are, the more rounded they are. When working with registers and memory addresses, you need bit-accurate numbers.

>This could be changed easily. I could add a number format modifier similar to what the old programmable pocket calculators had.

That is very interesting. My HP calculator uses "#" in front of a number to mark/store it as an integer. But I'm sure  it's a lot of work to add such a feature to TiniBasic. Also, who else would use it beside me? I don't know who are the typical TiniBasic users. Me, I am a hardware guy and low-level programmer. When I look at a MCU, I am not interested in it's speed or number of cores. It's the amount of intelligent peripherals that interests me. That's why I work so much with registers.

slviajero, 2025-03-05:
I just added a quick and dirty 64bit option. This should be able to handle 32bit integers in a floats as maxnum in 64 bit is 2^53. And as the GIGA is really fast, this should be no problem. I did not test this very much.

Just compile with HAS64BIT set in language.h.

easyprototype, 2025-03-05:
I just compiled a BASICFULL + HAS64BIT kernel for my GIGA. This config takes 12% of the flash memory instead of 10%.

Wow, just wow. I checked the new "integer" limit (9007199254740990 but errors creeps in below that) and it's huge!
My read and write command works, I can change my PWM and read my quadrature encoders directly.
My HEXTODEC command still works and I can even make it convert hexadecimal numbers bigger than 32 bits.

Who needs a number format modifier now?

slviajero, 2025-03-06:
Indeed! I was surprised how fast the GIGA is with 64 bit arithmetics. 

I wonder which version they put into the GIGA board. The Cortex M7 comes in three options, no FPU, single FPU and double FPU. From the speed I would say it is the latter.

slviajero, 2025-03-06:
Just checked. It has a double precision FPU.

easyprototype, 2025-03-06:
Wow. I might just use 64bits floating point for my project after all.

Is there a .bas file in the tiniBasic .zip that tests performance?

P.S.: I have expanded my little .bas file. I now test 32bits integer limit as well as 32bits float and 64 bits float. I don't think I can just call it test val and str anymore.

slviajero, 2025-03-07:
There is a set of benchmarks in examples/11stefans some of them are taken from a creative computing article summarized in the wikipedia page https://en.wikipedia.org/wiki/Creative_Computing_Benchmark#:~:text=The%20Creative%20Computing%20Benchmark%2C%20also,programming%20language%20on%20various%20machines.
very nice. 

Actually I mostly do this 

10 T=MILLIS(1): FOR I=1 TO 10000: A=5: NEXT: PRINT MILLIS(1)-T
RUN

It should be run as a program and not from the command line because otherwise it is not tokenized. 

An Arduino UNO with integer arithmetic runs this in 2000ms. The GIGA with 64bit float in 86ms when I tried it.

The performance bottleneck in BASIC is heap search of variables and memory access. The code tests both with about the same weight. I use this oneliner often to check if something with wrong after changes.

easyprototype, 2025-03-07:
I ran this morning creative.bas, rfbench1 to 8.bas and your 1-line test on 6 kernels:
ArduinoMega BASICINTEGER, BASICFULL and BASICFULL+HAS64BIT and
ArduinoGiga BASICINTEGER, BASICFULL and BASICFULL+HAS64BIT

I organized the result in a excel file. I can send it to you if you want.

While the difference between float and integer on the Mega is significative, there is almost no difference between 32bit int, 32bit float and 64bit float on the Giga. Long live the FPU!

I discovered that there is no 64bit float on the Mega. The compiler just compile in 32bits float.
I also noticed the software 32bits float seems to behave differently than the FPU 32bit float. I might be mistaken, though.

slviajero, 2025-03-07:
I am playing with this myself right now. Can you send me the excel? Strange the the mega has no 64bit.
gcc should do this right.

[11 more replies]