slviajero/tinybasic #65: Loop Error in TOWER.BAS

renorichard, 2024-07-07:
Hi Stephan,
I'm running on an Arduino Zero with the build downloaded 7/6/24. The only build option I enabled is
#define ARDUINOSD
I have the folder 14games copied to my SD card.
When I load & run TOWER.BAS it throws a Loop Error on line 1350 when I attempt to move the first disk.
I ported tower.bas to VB and it runs fine so this might be an issue in the runtime code.

slviajero, 2024-07-08:
Very bad. This seems to be an old bug. I tested it against 1.4 and it was also there in this version. The BASIC code uses a lot of GOTO out of loops. I fear that the loop cleanup code does not work as expected. I kill all inner loops once a loop is opened with the same variable. This may be different from MS BASIC in some situations. Will look at it this weekend.

slviajero, 2024-07-09:
I found the reason. Actually it is a bug in TOWER.BAS. When you look into the print subroutine, you find: 

1290 FOR V=1 TO T(K,J)
1300 PRINT "*";
1310 NEXT V

The variable V is reused although there is still an active loop with this name in the calling code. My BASIC strictly checks this and drops open loops to be GOTO save. BASIC is broken in this respect. Handling GOTOs out and into loops is not well defined. 

If you use another variable in TOWER.BAS, e.g. 

1290 FOR Y=1 TO T(K,J)
1300 PRINT "*";
1310 NEXT Y

the code works. I will checkin a new TOWER.BAS tonight. 

Still not sure if I want to change the basic code itself.

slviajero, 2024-07-09:
Changed TOWER code in repo.

renorichard, 2024-07-09:
Thanks Stephan - good eye!

Ported to VB I made the print routine a "Sub" so it had its own local
variable V and so "worked fine."

BTW I love your BASIC!

slviajero, 2024-07-09:
> 
> Thanks Stephan - good eye! 
> 
> Ported to VB I made the print routine a "Sub" so it had its own local 
> variable V and so "worked fine." 
> 

Exactly! This is really a question of making the loop variable and the loop local. 
I added CONT and BREAK to the BASIC very early to avoid the jump out of loop situation.
Unfortunately many of the old BASICs do this all the time. I found that the healthiest way of handling it is to delete all inner loops once you have a second loop with a different variable. This forces variables to stay local. But then, if you do what TOWER.BAS did originally, it means that you lose all the higher level loops and get a LOOP error. I could use a different strategy and look not only if the variable is the same but also the program location (i.e. the pointer here in BASIC). This would probably be more robust but may have other side effects. It looks like MS BASIC did this. 

> BTW I love your BASIC! 

Thank you. It is somehow more like C in some aspects. But then, I am a C programmer.

Best Regards and Thank you for the feedback,
Stefan
>