slviajero/tinybasic #66: TFT parallel display

Leizar06001, 2024-09-21:
Hi,
I am trying to make it run on a ESP32-8048S043 board, which contains a 800 * 480 display and a SD
So I set `#define ARDUINOILI9488`
Then in runtime, changed 
```
#ifdef ARDUINOILI9488
#include <Arduino_GFX_Library.h>
#endif
```
And also lower in the same file
```
#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
#define TFT_BL 2
/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
//Arduino_DataBus *bus = create_default_Arduino_DataBus();

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
//Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
    0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display tft = Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

/* Everything else stays the same excepted : (I removed the last arg) */
void dspprintchar(char c, uint8_t col, uint8_t row) {  if (c) tft.drawChar(col*dspfontsize, row*dspfontsize, c, dspfgcolor, dspbgcolor); }
```
Those are the correct settings for this display
The display seems to be initialized correctly, I can change its bg color, but for everything else it just stays black. 
Any idea ?

slviajero, 2024-09-26:
Hi, 

sorry for replying so late, but I have been moving house the last couple of days and was confronted with a few technical problems of a simpler nature.

If the display initialises correctly and can be accesses, which is obvious from being able to change the bg color, the most probable root cause can be color handling. BASIC maps colors using two functions converting from vga 16 colors to rgb. The code that does this is somehow messy because I had to integrate many boards with totally different color handling. 

You could try to call dspprintchar() with fixed arguments for the color first. Do really hard coded color. If this works you know 
that tft.drawChar does the right thing. Then you could look into color handling. Is dspfgcolor constructed right in the upstream code.

I can help you with this.

Leizar06001, 2024-09-28:
Hey ! 

Now worries, I know unfortunately there is a life outside of the code ^^

So I've managed to print something, first if I put `tft.drawChar(10, 10, 'X', 0xFFFF, 0);` in dspbegin, I got it on the screen. But putting a hard coded color in 'dspprintchar' didnt change anything

I noticed that the text was printed in the serial com and looking at the docs I found that with `#define STANDALONE` I can get it printed on the display, but following it there seems to be the prompt '>' printed endlessly, the display looks like this :
```
-------------------------------------------------------------------
|>  >  >  >  >  >  >  >  >  >  >  >  >  >                         |
|>  >  >  >  >  >  >  >  >  >  >  >  >  >                         |
|>  >  >  >  >  >  >  >  >  >  >  >  >  >                         |
|>  >  >  >  >  >  >  >  >  >  >  >  >  >                         |
|>  >  >  >  >  >  >  >  >  >  >  >  >  >                         |
|>  >  >  >  >  >  >  >  >  >  >  >  >  >                         |
|                                                                 |
|                                                                 |
|                                                                 |
|                                                                 |
-------------------------------------------------------------------
```

So, I wonder if maybe I didnt use the right defines, for now I have set those : 
#define DISPLAYCANSCROLL
#define ARDUINOILI9488
#define ARDUINOEEPROM
#define STANDALONE
#define ARDUINORTC

Second thing, It seems that the display is not fully used, is there somewhere else in the code where the resolution is defined ?

For information, I don't have any inputs connected to the esp, can it be messing with the display ?
I was thinking, if I can get your code working on my device, to try adding support for bluetooth keyboards later

slviajero, 2024-09-29:
Ok, I think now I understand the problem. If you go back to the original code and also leave STANDALONE unset then you should be able to access the display like this from the serial console: 

PRINT &2, “Hello Display”

This should provide the output on the display. You also should be able to change color and plot on the display.

In this mode, the display is the secondary output device and serial is the primary output and input device. 

Do you have a keyboard connected? Because only then STANDALONE makes sense. All other defines you use are ok

Leizar06001, 2024-09-29:
Yep it works, I don't have a keyboard plugged yet, I only have USB ones right now, I didn't think I'll ever need a PS2 one again !

My idea was to learn BASIC and for that I wanted to have a dedicated device that could only do that. At first I was looking at old computers but they are quite expensive now and have specific inputs/outputs. It only lately that I discovered projects for running this language on ESP. 

I also use a bluetooth keyboard everyday and I'd like to make this a portable device, so at the end it will be standalone, and when I set STANDALONE it behave as I described. Maybe the fact no keyboard is plugged make it believe I am pressing some keys ?

For the display size I only found now that there is `dsp_rows` and `dsp_columns` so that's good ! I'll make a specific define for this display instead of using the ILI9488 one

slviajero, 2025-06-22:
Closed fixed

closed 2025-06-22 by slviajero