slviajero/tinybasic #67: Event pin availability

EncomLab, 2024-10-03:
I am able to successfully use the EVENT command with pin 2 (as in the examples) - however if I change the pin to "3" I receive an Args Error.  Is EVENT limited to pin 2 only? 
This works as expected:
```
10 PINM 2,2
20 EVENT 2,2 GOSUB 1000
30 B=B+1
40 PRINT B
50 GOTO 20
1000 PRINT "TRIG"
1005 DELAY 100
1010 RETURN
```
This produces an Args Error:
```
10 PINM 3,2
20 EVENT 3,2 GOSUB 100
30 B=B+1
40 PRINT B
50 GOTO 20
1000 PRINT "TRIG"
1005 DELAY 100
1010 RETURN
> RUN
20: Args Error
> 
```

slviajero, 2024-10-03:
Your code does GOSUB 100 but the line does not exist in the second code part. It is GOSUB 1000 in the first.

Still, that means that the error message would be buggy because it should throw a line error.

EVENT should work with all interrupt pins. Can you check please if the line is the reason? I will look into the code then and improve the error messages. If the line is not the reason, I would need to check the interrupt code.

Best, Stefan

EncomLab, 2024-10-03:
Ugh - stupid to not catch that, but even when corrected the same issue remains.  What I am attempting to accomplish is to have two interrupts attached to motor encoders to provide PID control.  One interrupt works fine, but I need both to provide the reference for a PWM adjustment.
```
10 PINM 2,0
20 PINM 3,0
30 EVENT 2,1 GOSUB 100
40 EVENT 3,1 GOSUB 200
50 C=A-B
60 PRINT C
70 GOTO 30
100 A=A+1
110 RETURN
200 B=B+1
210 RETURN
> RUN
40: Args Error
```

slviajero, 2024-10-04:
Ok, that looks like a bug in the event code itself. I look into it this afternoon. We have a particulary rainy day today here in the Bavarian Alps. Gives me time for this.Sent from my iPad

40: Args Error

slviajero, 2024-10-04:
On my Arduino UNO it runs without any problems. I used the actual code in the repo not the release 2.0 code. 
There were a few problems with 2.0 concerning data types and run runtime.cpp module. 

Which version do you use? 

Best, Stefan

> On 3. Oct 2024, at 23:49, Encom Lab ***@***.***> wrote:
> 
> 10 PINM 2,0
> 20 PINM 3,0
> 30 EVENT 2,1 GOSUB 100
> 40 EVENT 3,1 GOSUB 200
> 50 C=A-B
> 60 PRINT C
> 70 GOTO 30
> 100 A=A+1
> 110 RETURN
> 200 B=B+1
> 210 RETURN

EncomLab, 2024-10-04:
This may be part of the issue - it's from the 1.144 release.  I extended that version with additional commands for motors and servos as I intend to use it for programming robotic platforms.  I have not tried extending the 2.0 version yet.

slviajero, 2024-10-04:
Ok, understood. Original 1.4 -> no error. 

I have a suspicion. The memory heuristics of 1.x tries to determine the best possible memory size and is very aggressive. It tries to grab as much memory as possible for BASIC. This can lead to heap overflows on small boards.

Set the MEMSIZE manually to a lower value. Like 100 bytes less than the heuristic would do. If this fixes the problem you have a heap size problem. In this case, in 1.4. look at freememorysize(). There is a section of the code that says 

#if defined(ARDUINO_ARCH_AVR) || defined(ARDUINO_ARCH_MEGAAVR) || defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_LGT8F) 
  int overhead=192;

overhead is the start overhead i.e. the heap size which is reserved for BASIC. Increase this a bit.

Best, Stefan

EncomLab, 2024-10-04:
Will test later today and update - thanks as always Stefan!

EncomLab, 2024-10-05:
Stefan it was the interrupt pin numbering - I am running this on a atmega1284 with custom pin definitions.  I changed the pin to another interrupt enabled pin and this works as expected now.

EncomLab, 2024-10-05:
Note to verify interrupt status of pins used for EVENT match the macro definition.

closed 2024-10-05 by EncomLab

slviajero, 2024-10-09:
Ok, understood. I will look into this and maybe add a better error message-

> On 5. Oct 2024, at 02:43, Encom Lab ***@***.***> wrote:
> 
>