slviajero/tinybasic #69: Exceeding program memory on the arduino uno

Oriebugg, 2024-10-29:
Is the Arduino Uno r3 still supported ? I've tried flashing basic1 onto my Uno but the IDE says it exceeds the program memory by 119%

slviajero, 2024-10-29:
yes supported which language and hardware settings do you use?Sent from my iPhone

slviajero, 2024-11-02:
Hi, just a few hints here. In version 1 all language settings are at the beginning of the .ino file. 

#undef  BASICFULL
#undef  BASICINTEGER
#define BASICSIMPLE
#undef  BASICMINIMAL
#undef  BASICSIMPLEWITHFLOAT
#undef  BASICTINYWITHFLOAT

An UNO runs with BASICSIMPLE, which is an integer BASIC. 

In addition to this setting, I recommend USEPICOSERIAL in hardware-arduino.h.

I corrected the default settings in the repository.

Oriebugg, 2024-11-03:
Hi! Thank you so much for your reply. I got it working now!! If I may ask, what's picoserial?