slviajero/tinybasic #72: Memory error when running on TTGO VGA32

coolcoder613eb, 2024-12-10:
I get a memory error whenever I assign a variable.

slviajero, 2024-12-10:
Which hardware.h and runtime.h settings do you use?Sent from my iPad

slviajero, 2024-12-10:
> Git diff output:
>
> diff --git a/Basic2/IoTBasic/IoTBasic.ino b/Basic2/IoTBasic/IoTBasic.ino
> index a1f331a..771d4e1 100644
> --- a/Basic2/IoTBasic/IoTBasic.ino
> +++ b/Basic2/IoTBasic/IoTBasic.ino
> @@ -128,8 +128,8 @@ const char spulse[]	PROGMEM = "PULSE";
>  #endif
>  /* DOS functions */
>  #ifdef HASFILEIO
> -const char scatalog[]	PROGMEM = "CATALOG";
> -const char sdelete[]	PROGMEM = "DELETE";
> +const char scatalog[]	PROGMEM = "DIR";
> +const char sdelete[]	PROGMEM = "DEL";
>  const char sfopen[]		PROGMEM = "OPEN";
>  const char sfclose[]	PROGMEM = "CLOSE";
>  const char sfdisk[]		PROGMEM = "FDISK";
> diff --git a/Basic2/IoTBasic/hardware.h b/Basic2/IoTBasic/hardware.h
> index b863138..16c2f10 100644
> --- a/Basic2/IoTBasic/hardware.h
> +++ b/Basic2/IoTBasic/hardware.h
> @@ -172,7 +172,7 @@
>  #undef AVRLCD
>  #undef WEMOSSHIELD
>  #undef MEGASHIELD
> -#undef TTGOVGA
> +#define TTGOVGA
>  #undef DUETFT
>  #undef MEGATFT
>  #undef NANOBOARD
> @@ -310,8 +310,8 @@
>   * ZX81 is not supported with different languages, I only have a UK one.
>   */
>
> -#define ARDUINOKBDLANG_GERMAN
> -//#define ARDUINOKBDLANG_US
> +//#define ARDUINOKBDLANG_GERMAN
> +#define ARDUINOKBDLANG_US
>
>  /*
>   * The hardware models.
> @@ -1071,3 +1071,5 @@
>  #else
>  #define FNLIMIT 128
>  #endif
> +
> +#undef DISPLAYCANSCROLL

MQTT seems to be broken. I need to look into this. Do you use it or does the error appear also without it?

coolcoder613eb, 2024-12-10:
I am not using it.

slviajero, 2024-12-11:
Which libraries do you use? Are they identical? I ran my usual test programs and they all were ok. 
Did you use the release code or the actual code in the repo right now? I tested against the actual code. 
If there is a bug in the release I will create a new release prestissomo.

slviajero, 2024-12-13:
> Repo code, and could you list all the libraries you use so i can find the versions?

the important ones are ESP32 core 2.0.7FabGL 1.0.9background info, the error you see (memory error on variable assignment) means heap corruption. this happens when the C stack memory overflows and poisons the BASIC memory.  on a esp32 this is very strange. if it is the repo code it could mean that my own dev tree is not in sync with the repo, i will double check this tonight Sent from my iPhone

coolcoder613eb, 2024-12-13:
I tried changing esp32 from 2.0.17 to 2.0.7, no difference.

slviajero, 2024-12-13:
I cross checked now the repo. The code is exactly the one I use on my development system. 
It compiles and runs correctly. Could it be that you accidentially made any other change? Anything that uses memory? 
Did you check FabGL. Ist it the newest one? 1.0.9?

Try to set MEMSIZE directly to cross check. You could try a 

#define MEMSIZE 8192

to radically reduce to 8k. 

I saw that you renamed some commands. This should be completely harmless. But maybe during editing you changed something?

slviajero, 2024-12-13:
PS: is this what you see after compile: 
Sketch uses 596897 bytes (45%) of program storage space. Maximum is 1310720 bytes.
Global variables use 30516 bytes (9%) of dynamic memory, leaving 297164 bytes for local variables. Maximum is 327680 bytes.

coolcoder613eb, 2024-12-15:
This is what I see:
Sketch uses 594729 bytes (45%) of program storage space. Maximum is 
1310720 bytes.
Global variables use 30644 bytes (9%) of dynamic memory, leaving 297036 
bytes for local variables. Maximum is 327680 bytes.
Memsize 8k doesn't help.

slviajero, 2024-12-15:
Does language.h specify#define BASICFULLif not, try this.Sent from my iPhone

coolcoder613eb, 2024-12-15:
It does.

slviajero, 2024-12-16:
This is indeed very odd. Could you please run the program 

10 ERROR  GOTO 35
20 DUMP HIMEM -20
25 SET 0,4
30 A=10
35 SET 0,0
40 DUMP HIMEM -20
50 PRINT USR(0,9),USR(0,13),USR(0,32)

for me?

coolcoder613eb, 2024-12-17:
[image]

slviajero, 2024-12-18:
Now it is clear where this comes from. You have set 

/* experimental features, don't use unless you know the code */
/* 
 * this setting uses the EEPROM as program storage 
 * The idea is to create a virtual memory layout starting from 0 with the EEPROM
 * from elength() and then adding the BASIC RAM to it. himem and top need to be 
 * handled carefully. 
 */
#define ARDUINOPGMEEPROM

which is not good on an ESP. It makes the EEPROM the core program memory which only works on the AVR. 
Unfortunately I have checked this in without any safeguard. Will correct this immediately. 

Please recompile with

#undef ARDUINOPGMEEPROM

coolcoder613eb, 2024-12-18:
It works!

closed 2024-12-18 by coolcoder613eb