slviajero/tinybasic #73: Develop this BASIC further to include more graphical capabilities.

OzHawk, 2025-01-18:
Hi Stefan,

Do you intend to develop this version of BASIC any further, or do you consider it finished?

If developing further, my request would be to implement some addition graphical capabilities.  Whether it was redefined character sets, loading and displaying bitmaps or more advanced like sprites, those features would allow it to be used as a stand alone computer more like the “boot-to-BASIC” home computers we grew up with.  Except that it would be possible to build you own “boot-to-BASIC” computer.

The computers of the 70s were OK, but they had very limited graphics until the late 70s.

slviajero, 2025-01-21:
Actually, I eas thinking about that a lot. But I havr not made up my mind which of the old basics to use as a model. Graphics is really complex to implement and a lot of work. The design has to be correct right from the start. Do you have any idea on that? Which basic was the best for graphics?Sent from my iPad

OzHawk, 2025-01-22:
Which of the old computers were best for graphics?
That’s a really tough question.  Back then, the graphics capabilities of a computer were heavily dependant  on the hardware used to drive the display.  For the late 70s, the Atari 400/800 probably had the best graphics based around the Antic chip.  Atari BASIC offered some capabilities, but when it came to programming the Antic chip it was still down to assembly code.  There were quite a few home computers based around the TI graphics chips.  While not as advanced as the Antic, they were still very capable.  BASIC on to Creativision even allowed programming of sprites for it.  However, that was the slowest version of BASIC ever.  The Vic20 offered no sprite capability in the Vic chip, however this could be achieved in software by redefining the character set.  I think being able to redefine the character set would be a minimum, and then consider how you would set the colours of the individual characters on the screen.  I’m unsure of the chip used in the ZX Spectrum, but it was capable of amazing things even though it only allowed the foreground and background of each character on the screen to be set from a limited palette.
In modern BASICs, MMBASIC has implemented a full suit of graphics capabilities from within BASIC.  BBC BASIC also has an interesting way of expending the graphics capability of a machine from within the existing BASIC framework.

slviajero, 2025-01-27:
> I think being able to redefine the character set would be a minimum, and then consider how you would set the colours of the individual characters on the screen. I’m unsure of the chip used in the ZX [...]

Redefining a character set is relatively easy if the underlying hardware supports it. I have not yet implemented it because other things have been more important. Most displays that BASIC support use software fonts anyway, this means one would simply overlay a certain ASCII range (typically 224 upwards) and use alternatives fonts. Setting different colors for each character is already implemented and can be done with the COLOR command or by sending an VT52 sequence to the terminal. I have implemented the full GEMDOS/TOS VT52 extensions into the display driver code. This is, however, not tested very thoroughly. 

Biggest obstacle currently is the hardware. There are so many different hardware drivers and capabilities. 

Can you write brief user stories for the features you would need? Like just one or two lines. I would then look into them. At the moment I am busy with new hardware but afterwards I will look into graphics.

One thing I look into right now is an additional output channels &3 that can display text and sprites anywhere on the display i.e. outside the lines/columns structure of the display driver. I need this for Epaper display.

OzHawk, 2025-01-27:
I would look at emulating a video dosplay processor that was commonly used by various computer back in the day.
Selecting the MOS 6560 (VIC), MOS 6567 (VIC-II) or ANTIC will limit interest to only the users that had those computers. (Admittedly that was a LOT of users), but others are keeping those users satified with modern remakes.
I would look at emulating the TMS9918, that was used in a series of computers and consoles.  Or one of fthe improved versions.
A lesser Video Interface Controller to emulate would be the Motorola MC6847.  It was used by some cheap low end home computers.  A lot less capable though.
One of the nice features of the VIC and VIC-II chips was the ability to change the colour memory without changing any other memory.
If you have enough memory, you would want to implement at least two framebuffers.  The amount of memory available limits the resolution of the framebuffers.  A common resolution 320 x 192 (different to the TMS9918) giving a character screen of 40 x 24 8x8 characters.
I'll consider some user stories to help you with your design.

EvilDaveCanada, 2025-01-27:
I've subscribed to this thread and just noticed the last few comments.
I used to write custom software in a Business Basic professionally. The
language is still in use and maintained by Basis in New Mexico.
The original language was known as MAI Basic. When MAI went out of
business, Basis created a new language based on it. This new language was
called BBx, which stood for Business Basic Extended. I have complete
command manuals for it and a few newer versions of it.
Basis later developed a version where the BBx code (which was an
interpreter) could be compiled in to Java.
BBx used a slightly customized version of the UNIX Termcap file to handle
I/O to monitors, serial terminals, MSDOS command line, printers and
plotters. They needed a modified version of Termcap because there were
things they needed that were not part of Termcap when they developed BBx.
My suggestion is to use the most up-to-date version of Termcap now. But
BBx's major use was it's special auto sorting files.
I have always wanted to create an open source version of it and the
complete file structures are detailed in one of my old manuals.
There are also a few attempts at a Business Basic here on GitHub that I
follow. They have all sputtered to an end. I believe the cause of these
failures was due to the lack of a complete understanding of how the
language handles their special files. The only thing I would want to do is
instead of compiling into Java, compile into Python and possibly ASCII C as
well.
I would enjoy spending time going over things and sharing my knowledge as I
spent from 1981 through to 2003 working in this language. I used to be the
Systems Administrator for a major garment manufacturer as well as writing
and maintaing accounting software for law offices & a vertical blinds
manufacturer. My last major job involved helping convert a huge payroll
system that ran on an old mainframe on to three(3) linked UNIX systems all
running a payroll running on BBx. Please feel free to contact me through
GitHub.

slviajero, 2025-02-01:
just had a look at the business basic manual. wow! i found out that I actually had access to one of these systems. we got one that was decommissioned in the late 80s and played around with it.the basic had all the file manipulation features that you describe. and it could adress a terminal.problem with termcap would be to port it to mcu platforms. i used the vt52 commands for my basic because they are easy ro implement. as to graphics, I think the business basic had none? but i may be wrong.Sent from my iPhone

slviajero, 2025-02-02:
I looked into the old literature a. It. VGA or EGA would be a good standard. Most DOS stuff would work on it. But for gaming they were poor.Sent from my iPad