slviajero/tinybasic #74: It's not a issue. What I use for STM32F4 Board ?

moahrs, 2025-02-16:
I apologize in advance because I can't find a place to talk to the developer. 

I read everything but I didn't find any direction. Which of the basic2 subfolders can I use for my STM32F4 board project? It has 1MB of flash and 128KB of ram. 

I think it would be the arduino one, but I'm not sure. I compile in the stm32cubeide and not in the arduino ide. Thanks.

moahrs, 2025-02-16:
sorry.. sorry.. very sorry... i found in sources... IOT....

closed 2025-02-16 by moahrs

slviajero, 2025-02-17:
> moahrs
>  left a comment
> (slviajero/tinybasic#74)
> sorry.. sorry.. very sorry... i found in sources... IOT....

Just for info, played around with the blue pill STM32F103 boards now. They work perfectly with BASIC. Enjoy! 
If you find any issues with the F4 please report them.