slviajero/tinybasic #78: Suggestions (for more apps)

FreeLANMan, 2025-05-18:
First of all, congratulations on the excellent work. The compatibility with so many different devices is really impressive.
And the documentation is very good, but I missed a document explaining in a summarized way the syntax of this BASIC (like this one: https://www.infinite-loop.at/Power20/Documentation/Power20-ReadMe/AA-VIC-20_BASIC.html).
I also think it would be interesting to have a command, used at the beginning of the code, that informs the interpreter that this code is in a specific dialect (Palo Alto, Apple Integer, Dartmouth). In this case, my interest is in the easy use of existing applications (and in the easy development of new ones using AI).
And thank you for the excellent work that can favor the development of low-cost and low-power computing.
(And also if you want to take a look at the system I created for a computer based on a microcontroller: 
https://github.com/FreeLANMan/PCEmulatorPlus

slviajero, 2025-05-22:
First of all, congratulation to your own project. I like this and will definitely have a look at it. Also, the idea with the syntax summary is good. I look at it. Originally MANUAL.mp was meant to be that but it has grown a lot. 

The command to change the settings is there. It is SET 22

SET 22, "Microsoft" makes the interpreter a MS BASIC (with some restrictions)
SET 22, "Apple1" a Apple 1 BASIC 
SET 22, "Palo Alto", a Palo Alto BASIC. 

These settings only affect the way certain features work. The first one is the most powerful, as it makes the string commands compatible with MS BASIC and changes the way RND and array indices work to standard.

sapbotgit, 2025-05-22:
There is no need to use other dialects since TinyBASIC is specific standart. You can just develop for it, and it will be easy to port it to Apple II or Commodore 64. https://en.wikipedia.org/wiki/Tiny_BASIC